OctoPrint/OctoPrint #814: [Request] Poll system data from api

wolfspyre, 2015-03-15:
Hi there,

I was wondering if you thought it might be reasonable to make some telemetry data about the host running OctoPrint available via the API (likely requiring auth token)

I was thinking things like the hostname, memory total/free/ network info, cpu load, etc.

I looked over the [API Documentation](http://docs.octoprint.org/en/master/api/index.html), and didn't see this listed.
I also didn't see a similar request in already closed issues. (but I could be blind, please excuse me if I missed it)

While this could be considered out of scope for OctoPrint, and that this information should be enabled via a plugin; I could also make the argument that as a general indication of overall system health, the data is useful enough to warrant discovery via the web UI anyways. If that assertion rings true, then making the data natively available via the API makes perfect sense to me.

 My motivation for the request is that I am hoping to incorporate some additional system health data into the [dashboard](https://github.com/wolfspyre/dashing-octoprint) I'm putting together, and having this available would mean I wouldn't need to poll the OctoPrint host via some other means.

NI2O, 2015-04-19:
+1 please!!

It would be useful to be able to poll other data too such as the temps and the %complete etc. That info could be painted onto the webcam picture for general display on other devices without needing to log into the printer interface.

foosel, 2015-06-04:
OctoPrint is supposed to be platform agnostic, and getting the host system's health data is probably not going to be possible in a platform agnostic way => plugin ;)

JeffyW, 2017-06-16:
FYI, I too was looking for a similar thing, so I I dug into it a bit.  In particular, I was interested to see how much of an impact webcam streaming had on the performance as well as perf details when the terminal would start reducing functionality.

I looked into how [NavbarTemp](https://github.com/imrahil/OctoPrint-NavbarTemp) did this with temperature and wanted to extend it.  While I figured that'd be an OK starting point, I came up with dozens of ways to display the data in my head.  At the end of the day, I decided a good model would be to make a single plugin that simply published data.  That would allow other plugins to display the data in useful and relevant ways.  It also meant if I had 2 plugins that both wanted similar data (such as a NavBar for fullscreen and potentially some mobile alternative), I would get a consistent refresh time and wouldn't have multiple background threads doing the same work.

Right now, I have CPU% (both overall and per core) and Memory (%, total, available, free, used) implemented using psutil (which should work on most platforms).  I'm also publishing temperature for linux platforms using `/sys/devices/virtual/thermal/thermal_zone0/temp` or specifically for RaspberryPi `/opt/vc/bin/vcgencmd measure_temp`.  I intend to add a bit more info, such as disk space and cpu frequency (for throttling).  I may also try and add some APIs that can be called to get static information (hardware & cpu) as well as possibly some for getting top processes by cpu / memory.

The plugin that publishes the stats is [ServerStats](https://github.com/JeffyW/OctoPrint-ServerStats).
Here's a sample screenshot of it in action:
[image]
While debugging, I'm printing the `data` sent to `self.onDataUpdaterPluginMessage` so you can see the type of data that's available.
The UI is from my fork of [NavbarTemp](https://github.com/JeffyW/OctoPrint-NavbarTemp/tree/serverstats) just used for testing.

foosel, 2017-06-19:
@JeffyW Nice work! Might make a nice addition to the plugin repository :)

jurcja1, 2017-08-26:
Hi, I tried to install plugin via URL https://github.com/JeffyW/OctoPrint-ServerStats/archive/master.zip
During install process I got this error:

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-C0kUbc-build/
Error!

Could you please help me?
Thanks!

LazeMSS, 2021-03-31:
There are plugins that expose these data and furthermore there are no 100% sure way to include it - but the best way would be to use the PSUtil which is already included in octoprint - making an API for all this almost overkill - the PSUtil has it all and is used by plugins - I would think developer should use that - that way it works for other hardware platforms.

https://github.com/giampaolo/psutil

closed 2021-03-31 by LazeMSS