OctoPrint/OctoPrint #815: Shutdown printer

jrhrdk, 2015-03-15:
When the printer gets its power from an atx psu, the psu can be turned on by connecting ps_on to ground - it would be nice if octoprint could control a switch which could break this connection, so the printer gets turned 100% off. And while we are at it: it would be nice if it was possible to connect a switch which would be activated when the filament went out to the raspberry, so the printer would stop when there were no more filament!

closed 2015-03-15 by GitIssueBot

nophead, 2015-03-15:
For the first issue it can, see http://hydraraptor.blogspot.co.uk/2014/06/lights-camera-action.html.

You can probably do the second issue with a python script reading the gpio in the background and posting to octoprint's api to force a pause or stop.

Grogyan, 2015-03-15:
Probably won't be able to control this line directly with the PI as many
users are now using the GPIO for small touch displays.

jrhrdk, 2015-03-15:
Nice to have different options thou.. (o:

Regards
John Rask