OctoPrint/OctoPrint #819: "Cannot capture image, image number is unset" causes rendering to fail

guysoft, 2015-03-20:
1. What were you doing?
Printing a timelapse with the setting: on Z axis change

2. What did you expect to happen?
Get a lovely timelapse of the print from start to end

3. What happened instead?
I got a timelapse of part of the print

4. Branch & Commit or Version of OctoPrint:
devel

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure): Rigidbot 

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure): Firefox

7. Link to octoprint.log on gist.github.com or pastebin.com:

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):

The logs says at the cut of the resulting video

    2015-03-20 11:14:53,602 - octoprint.timelapse - WARNING - Cannot capture image, image number is unset

NOTE:
Current workaround is to copy all the images from the ~/.octoprint/timelapse/tmp/ folder and render manually with:
ffmpeg -r 120 -qscale 1 -i tmp_%05d.jpg output.mp4
(you can change r for different framerates)

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

None

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

None

I have read the FAQ.

foosel, 2015-03-20:
Please add either the full version string or branch AND commit. Just "devel" is not specific.

guysoft, 2015-03-20:
Updated: 1.2.0-dev-590-g60882d9 (devel branch)

foosel, 2015-03-23:
Did the "image number not set" log line happen at the end of your print, in the middle, at the beginning? Does it happen every time?

guysoft, 2015-03-23:
image number not set happen during the print (at least the timestamp says so). 
Does not happen every time. Has a really high probability (close to 1) when doing long prints of 5+ hours.

murphy2712, 2015-06-29:
I'm using v1.2 stable with the "Time" mode and have the same error.
The image number is not incremented:

pi@octopi ~/.octoprint/timelapse/tmp $ ls -lh
total 48K
-rw-r--r-- 1 pi pi 24K Jun 29 13:59 tmp_00000.jpg
-rw-r--r-- 1 pi pi 24K Jun 29 16:47 tmp_00001.jpg

It keeps writing again and again on the same file "tmp_00001.jpg".

ss89, 2016-11-20:
i happen to see the same, though the ip of my webcam server changes a lot => could be the reason of this problem (on my side).
Possibly a capture problem with mjpgstreamer/motion/fswebcam/etc.?

foosel, 2019-11-05:
Haven't heard of this happening again for almost 3y and never saw it myself. Considering it solved.

closed 2019-11-05 by foosel