OctoPrint/OctoPrint #822: [Support] Raspberry Pi Camera

MrRebel, 2015-03-20:
Recently I was trying to stop the message of no camera detected as I had no camera, so I turned off the camera in the raspi-config, and it stopped. Later I got my hands on a raspberry pi camera module. I turned camera settings back on in raspi-config and expected it to work properly. However it did not and all I get is a small box in the control window of the Octopi.local page showing that there is no image. I did not change any other files though. Do I need to?

closed 2015-03-20 by GitIssueBot

foosel, 2015-03-20:
Please, is it really too much to ask to actually read the guidelines you are repeatedly referred to? :(

They would tell you that support requests are to please be directed at either the mailing list or the g+ community.

MrRebel, 2015-03-20:
I'm very sorry!
My second post used the 5 questions used in the snapshot shown in the GitIssueBot but it was still closed. But I asked my questions without researching what I was supposed to. In the future I will try to figure exactly what to do if I run into a problem.
Sorry again!