OctoPrint/OctoPrint #825: GCode Viewer Size Override Not Working

eboston, 2015-03-25:
1. What were you doing?
Selecting GCode Viewer tab with a large print file.

2. What did you expect to happen?
After getting the warning about  the file size and clicking the visualize regardless of size button, for the GCode Viewer to show the visualization.

3. What happened instead?
The blank visualization screen is shown then it goes back to the warning message.

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-668-g3ebd54d (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
Printrbot Simple Metal

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Chrome v41.0.2272.101 m

7. Link to octoprint.log on gist.github.com or pastebin.com:
Nothing was shown in the log

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):
Nothing shown in the terminal tab

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
No errors were displayed in the javascript console.

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
N/A

I have read the FAQ.

This same behavior is seen on my mobile devices and desktop browsers.

cscott, 2015-04-02:
I'm seeing this, too.

Grifplex, 2015-04-08:
[image]
I'm also seeing this as well as selecting 'Zoom in on model' does not work since I went to Octoprint v1.1.1-32-gd974ab0

foosel, 2015-05-04:
@eboston @cscott I just tested against current `devel` as well as the version reported in the original ticket. I can't reproduce it. Can you give me any more info? E.g. does it always happen or only with some models, some file sizes etc? Any pattern at all?

@Grifplex only one issue per ticket please.

eboston, 2015-05-04:
@foosel Not sure why you can't reproduce it as I can consistently on all PC browsers and Android Chrome and Dolphin. What size is the file you are trying it with? How much RAM is in your system? I can send you a gcode file this happens to me with when I get home tonight.

eboston, 2015-05-04:
As for file size, it happens with files > 20mb.

foosel, 2015-05-04:
I tried by setting the threshold for this message down from 20MB to 1MB, then tried loading a 9.5MB file. No issue. So maybe it's caused not by the actual mechanism in place but rather indeed by file size. A test file would help, thanks in advance.

eboston, 2015-05-04:
@foosel I sent you an email with a link to a file that causes the problem on my system.  It is 22.7MB as reported by OctoPrint when loaded.  Just for reference.  Running OctoPi which has OctoPrint Version: 1.2.0-dev-758-g2ce40c4-dirty (devel branch).  I am using a Raspberry Pi 2 Model B+ (newest version).  My PC is running 64 bit Windows 7 Home Premium SP1 with 8GB or RAM.

foosel, 2015-05-05:
@eboston hm, mail hasn't made an appearance yet (also looked into the spam folder) :( can you upload it somewhere (drive, box, dropbox, anything like that) and post the link?

eboston, 2015-05-05:
It was a link to a Dropbox file. I hate posting it publicly,  but here it is again... 

https://www.dropbox.com/s/94ppp8xqb9hckvk/Legs.gcode?dl=0

foosel, 2015-05-05:
Ok, just downloaded it and now I have it, you can delete it so it doesn't stay public longer

foosel, 2015-05-05:
Hm... I still can't reproduce it. Will try from a Pi (although since it's client side it shouldn't actually make a difference, but if it's some odd network problem...)

foosel, 2015-05-05:
Good news, on the Pi it is reproducible

foosel, 2015-05-05:
Possible fix committed, can you verify?

eboston, 2015-05-05:
I'll update tonight after work.

foosel, 2015-05-05:
Awesome, thanks

eboston, 2015-05-05:
Pulled the current version and verified that it is working now.  It actually generated the view for that file which it has never done before.

foosel, 2015-05-06:
Perfect, closing then!

closed 2015-05-06 by foosel