OctoPrint/OctoPrint #826: Manual Extrude not working

net127, 2015-03-27:
I am unable to manually extrude on 1.1.1-27 with a Solidoodle 3 (stock configure) and Lubuntu 14.04. In the "Select Tool.." drop down, there is a "Select Hotend" but it does not select and pressing the "Extrude" button does nothing. XYZ buttons work. Printing works.

Excellent software!

closed 2015-03-27 by GitIssueBot

mohankumargupta, 2016-01-31:
I am having the same issue . Using octopi and a prusa i3. Like the original poster mentioned, all the other printing functions work fine, but when you select hotend from dropdown,  the dropdown reverts to select a tool.
extrude/retract don't work  (chrome browser). Is this a cura related setting . Where is the tool list populated from?

shaiss, 2016-03-29:
any updates on this one as I'm having the same issue with repetier firmware.

foosel, 2016-03-29:
No full bug report, no chance on an update. Same reason the original ticket was closed really. We don't have telepathic capabilities, we cannot solve issues on thin air. So add information to #1301 (which was just opened today) - [and make sure to include all required information so we actually can help](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

Nvigr8, 2016-09-04:
Could it be something as simple as not having your temps up?  It seems the software won't allow one to extrude till the set temp is met.  Or something to that effect.

markwal, 2016-09-05:
The select tool button just allows you to send a T0, T1, T2 or what not to the printer.  It doesn't know what tool the current printer has selected because the normal status doesn't necessarily say.  So I can see why you might expect the button to have feedback, but it doesn't.  You just click the tool you want and it sends that tool selector gcode to the printer (you can check the terminal window).

Extrude/Retract commands the printer to move the extruder, it doesn't check to make sure the temperature is sufficient to do it, but your firmware might.  On many printers if the filament isn't hot enough, what you'll hear is clicking as the extruder stepper motor skip steps as it fails to push the cold filament.

SomeKittens, 2018-01-20:
If anyone's still reading this, the solution suggested by @Nvigr8 worked for me (I'm on a Lulzbot Mini).  I had thought I had set the hotend temp but it was still off.  As soon as I let it heat up, I was able to extrude just fine.