OctoPrint/OctoPrint #829: Unicode serial crash with VIRTUAL printer

jminardi, 2015-03-30:
1. What were you doing?
Disconnecting and reconnection to a virtual server after canceling an in-progress print. The in-progress print was seemngly hanging when the temperature was almost at the set point so I canceled the print, disconnected and attempted to reconnect.

2. What did you expect to happen?
I expected octoprint to connect to the virtual server without error.

3. What happened instead?
Octoprint was not able to connect to the server and instead displays an error

4. Branch & Commit or Version of OctoPrint:
devel 2fa0673e0b824e9c9d70bbe2d32d6599d3435d09

5. Printer model & used firmware incl. version
VIRTUAL printer

6. Browser and Version of Browser, Operating 
Chrome Version 41.0.2272.104 (64-bit) on OS X

7. Link to octoprint.log on gist.github.com or pastebin.com:
http://pastebin.com/99XZYFXT

8. Link to contents of terminal tab or serial.log on 
http://pastebin.com/ky7Q7wcb

9. Link to contents of Javascript console in the browser 
Nothing out of the ordinary in the javascript console

10. Screenshot(s) showing the problem (if applicable - always
N/A

I have read the FAQ.

Error message:

```python
2015-03-30 00:37:34,937 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/Users/jack/code/voxel8/OctoPrint/src/octoprint/util/comm.py", line 791, in _monitor
    line = self._readline()
  File "/Users/jack/code/voxel8/OctoPrint/src/octoprint/util/comm.py", line 1254, in _readline
    self._log("Recv: %s" % sanitize_ascii(ret))
  File "/Users/jack/code/voxel8/OctoPrint/src/octoprint/util/__init__.py", line 392, in sanitize_ascii
    return unicode(line, 'ascii', 'replace').encode('ascii', 'replace').rstrip()
TypeError: decoding Unicode is not supported
```

jminardi, 2015-03-30:
The following patch seems to fix the issue but I am not sure if this is the correct approach or if it is just putting a bandaid on the problem.

```diff
diff --git a/src/octoprint/util/__init__.py b/src/octoprint/util/__init__.py
index c16dbee..c0915b7 100644
--- a/src/octoprint/util/__init__.py
+++ b/src/octoprint/util/__init__.py
@@ -389,7 +389,7 @@ def silent_remove(file):
 
 
 def sanitize_ascii(line):
-       return unicode(line, 'ascii', 'replace').encode('ascii', 'replace').rstrip()
+       return unicode(str(line), 'ascii', 'replace').encode('ascii', 'replace').rstrip()
```

foosel, 2015-03-30:
I just pushed something that might solve this... I cannot reproduce the issue at the moment (I guess it has something to do with some odd filename in your virtual printer's SD card folder that causes this problem, but I have some umlaut heavy files for testing there myself and never ran into this issue), but I just made sanitize_ascii now only convert the line to unicode in the first place if it's still a str (and raise an error altogether when it's neither unicode nor str).

Can you test if that solves the issue for you? Also, what Python version are you running (maybe that's the problem here).

Oh, and - judging from the path names in the log and your own name, are you working on getting OctoPrint to work with the Voxel8? ;)

jminardi, 2015-03-30:
Hey @foosel thanks for the quick fix. And yes, there is a good chance we will be allowing our customers to use OctoPrint with the our printer at Voxel8

This seemed to fix the immediate problem however now I am getting a stall in the print I am trying to test. Once the temp reaches the set value the temp graph freezes and I get a fast stream of incorrect line errors.

Here is the log file:
http://pastebin.com/K5fdRkVn

And here is the gcode I am trying to test with:
http://pastebin.com/kzauk9p4

I am not sure if this is related, I can open a new issue if you think its separate.

foosel, 2015-03-30:
I'll need a serial.log here to see what's going on right before the freezing. My guess though is that it probably is not related. This is still against the virtual printer?

jminardi, 2015-03-30:
My serial.log is generated but empty. This issue really isn't blocking me and as you said probably unrelated. I will close this issue and open another one if I have some more reproducible steps.

commit 79336ca1 (2015-03-30): Fix: Improved handling of data coming in from the printer

Hopefully closes #829
closed 2015-03-30 by jminardi