OctoPrint/OctoPrint #831: How to implement PAUSE at layer == x control for multicolor prints with single extruder?

bobgatpdx, 2015-03-31:
Hello,
I'm using Octopi with my Printrbot Simple Metal - works great!
It would be very useful to be able to automatically pause the print when it reaches a certain layer so I can change the filament color.  Can anyone suggest a good way to implement this feature?  Seems like it would be useful to many folks.  How difficult would it be to create a custom control for this?
thanks,
Bob

closed 2015-03-31 by GitIssueBot

eboston, 2015-03-31:
@bobgatpdx,I implement it by using Repetier when I need to do that kind of printing.  There is a plugin that is suppose to support this feature, but it doesn't work and doesn't look like it will be fixed any time soon.

foosel, 2015-04-01:
Insert an M0 at the location of the gcode by the slicer. Does the same
thing as @pause in repetier from my understanding. Octoprint will pause the
print when it sees that, can then be resumed when done.

bobgatpdx, 2015-04-01:
Thanks Gina.  Yes, I tried this and it works OK.

I used the "pause at height" plug-in for the Cura slicer.  It inserts the following extra lines into the gcode.
These lines move the extruder out of the way and pauses the print.  Then you can manually change the filament.  After changing the filament, then you can click the PAUSE button on Octoprint and the print restarts. 

So this works OK, but it would be really slick if this feature could be implemented on the Octoprint side by doing the same thing when the layer count gets to the desired point.  Using the Cura plug-in is very cumbersome and it's hard to figure out which layer to insert the code into.  I'd be happy to work on this, but I'm new to Octoprint.  Any advic
[image]
e would be appreciated.

best regards,
Bob
======================================

gcode inserted by "pause at height" plug-in to Cura slicer:
;LAYER:13
;TYPE:CUSTOM
M83
G1 E-5.000000 F6000
G1 Z15 F300
G1 X150.000000 Y150.000000 F9000
M84 E0
M0
G1 E5.000000 F6000
G1 E-5.000000 F6000
G1 X91.957000 Y37.205000 Z2.900000 F9000
G1 E5.000000 F6000
G1 F9000
M82
G1 F2400 E4609.49607
G0 F4200 X82.276 Y56.316 Z2.900

discojon, 2015-04-01:
Don't think that you can copy and insert the above code and all will be peachy.  Parts of that code are specific to that print.  I struggled with this for a while myself and this is the current pause code I am using to change filament on my Taz 5.  This is inserted after the last layer you want of a certain color.  This is a generic code that should work with all files.  

;Pause Code
G91				;Set Relative Mode
G1 E-5.000000 F500	;Retract 5mm
G1 Z15 F300			;move Z up 15mm
G90				;Set Absolute Mode
G1 X20 Y20 F9000		;Move to hold position
G91				;Set Relative Mode
G1 E-40 F500		;Retract 40mm, this can be commented out if you just want to pause to insert magnets or something.
M300                       ;Beep (marlin)
M0				;Idle Hold
G90				;Set Absolute Mode
G1 F5000		;Set speed limits, depending on slicer this can be set to your initial mm/min speed or it can be used for the following move only then the next layer will set the speed
G28 X0 Y0		;Home X Y
M82				;Set extruder to Absolute Mode
G92 E0			;Set Extruder to 0

bobgatpdx, 2015-04-01:
Hi @discojon,
You are absolutely correct.  I was just showing the code that Cura generated for a particular print I was doing.  You're gcode example is much more universal.  

Ideally, I'd like to create an Octoprint FILAMENT_CHANGE plug-in that does the following:
1) Before starting the print, user enters this info in Octoprints Web UI: layer# to stop at, pause_gcode_file (up to M0 in your example), restore_gcode_file (after M0 in your example)
2) Plugin is enabled and print is started
3) Upon reaching the proper layer, the pause_gcode is inserted and the print is paused.
4) User manually changes the filament
5) User presses PAUSE on Octoprint to restart the print.  The restore_gcode is executed and then the print resumes.

This would eliminate editing gcode files for each new print.  All I need is some advice about how to do this within the Octoprint plug-in system - I'm new to Octoprint.
- Bob

shadycuz, 2018-01-07:
@bobgatpdx Did this ever come about ?

tcurdt, 2018-09-09:
@shadycuz I was wondering the very same thing. I found https://github.com/OllisGit/OctoPrint-PauseAt but that doesn't seem ready for prime time yet.

tcurdt, 2018-09-09:
Another option seems to be:

https://plugins.octoprint.org/plugins/multi_colors/
https://github.com/MoonshineSG/OctoPrint-MultiColors

discojon, 2018-09-09:
The multicolor plugin does exactly what I was looking for.  If you have a recent version of Marlin just have it insert a m600.  The specifics of m600 can be set in your fw if standard settings don't work.