OctoPrint/OctoPrint #833: [Request] Make it more difficult to accidentily delete files

jacksonliam, 2015-04-02:
1. What were you doing?
Trying to 'Load' a file, but clicked delete button by mistake

2. What did you expect to happen?
Confirmation or button not so tiny and next to open button

3. What happened instead?
Button is tiny enough with mouse let alone with touch!

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-660-g3299dbd (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
N/A

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Firefox 37.0, Also Chrome etc

7. Link to octoprint.log on gist.github.com or pastebin.com:
N/A

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):
N/A

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
N/A

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
[image]

I have read the FAQ.

Ideas I had:
- Delete confirmation
- Delete button only when file details expanded.
- Sacrifise the UI space to show buttons row as wide as the full file box (are filesize and uploaded date needed? can they not be hidden in expandable details e.g. `ls` vs `ls -l`)

Cheers! Liam.

REulberg, 2015-04-03:
I agree -- a pop up confirmation would be helpful both here and on the time lapse panel

foosel, 2015-04-03:
See also #748 and #687

braddo99, 2015-04-05:
Please no confirmations, especially when these files are by definition sourced from somewhere else.

Best practice is to put the delete button on the other side of the dialog away from the other buttons. (like the classic OK / CANCEL separation)

Also all of the buttons are super small, they could be significantly bigger and have the print entry occupy the same space if the "uploaded a few seconds ago" were shorter and in the same line as the file name, and the file size info were put into the detail dropdown.

I'm not sure a Load and Print button is required - it's convenient, but a bigger commitment to directly print.

Personally, I would prefer that when a file is uploaded, it's immediately opened and ready to print unless something else is already printing.

robseee, 2015-11-23:
i vote for a deleting confirmation

BloodyIron, 2024-05-29:
So 9 years later and we still don't have a "ARE YOU SURE YOU WANT TO DELETE THIS FILE YOU MIGHT CARE ABOUT?!?!?". Really????

foosel, 2024-05-29:
9 years later this issue has gotten next to no traffic, no upvotes, only three opinions apart from your own, and absolutely no PRs whatsoever. So apparently it is not a high priority to OctoPrint's user base.

With that being said, I acknowledge there is room for improvement here, but that involves something more complicated than just slapping a confirmation dialog on it and call it a day, as that will break the bulk delete workflow of many people out there. It will likely be addressed another way by a forthcoming bundling of the FileManager plugin.

With that being said, you need to work on your tone and attitude. This is an open source project. If something annoys you, you are welcome to contribute in a constructive and friendly way to improve on it. This kind of tone you showed is however not welcome here, no matter how frustrated you are at something that went wrong somewhere. [You should read up on the social contract of open source](https://snarky.ca/the-social-contract-of-open-source/).

BloodyIron, 2024-05-29:
> 9 years later this issue has gotten next to no traffic, no upvotes, only three opinions apart from your own, and absolutely no PRs whatsoever. So apparently it is not a high priority to OctoPrint's [...]

I work with countless Open Source projects and frankly I think the lack of tolerance for people being upset is a mistake. I did not say anything that was offensive, nor an attack on anyone, or anything like that. I conveyed a surprised impression that something as rudimentary as this was not in-place, despite this GitHub Issue being open for so long.

I don't need to check my attitude and I am well versed in how to actually be respectful in these open source and other spaces, and what I said was not unreasonable to any degree.

Considering there is zero undelete (as far as I can tell) and zero protections against accidental deletion of files, I think the degree of my expressed surprise is warranted and reasonable. I have not yet slipped up and hit delete on the wrong file,  but as a human I very easily can. With how much time I myself put into each of my gcodes, it would be a huge problem for me to accidentally fat-finger deleting a file without any "are you sure?" dialogues. I speculate that there are an endless supply of other people that put a lot of time into their gcode too.

It is extremely commonplace (all applications typically) to have a "are you sure?" type dialogue when deleting files. Literally every OS has one sort of mechanism or another to do it (whether it's sudo, GUI prompts, whatever). So again, the surprise to me to find nothing like that exists in OctoPrint even after 9 years is completely warranted, rational, and reasonable.

You may not like that I am concerned, or that I expressed surprise in the duration, but that's what I had to say and in no way was it disrespectful, an attack on someone, or unwarranted. Frankly if you can't handle what I said, you need to grow a tougher skin.

I also want to point out that to contribute to Open Source projects, it doesn't _require_ that you contribute code. I myself contribute endless supplies of bug reports, feature requests, and more, where I exhaustively test many different suites of FOSS and report as useful info as I can (often with deep insights) because that is the area I can contribute to significant degree. I know this isn't exactly a bug report thread, but bringing attention to this topic (which was my intent with what I said) has its value too. To dismiss that is in my opinion foolish and short-sighted. And to then cite gospel that I "need to work on my tone and attitude" (your exact words just now) is not actually seeing what this whole thing is about, but instead preferring to tell me to put out my wrists and ignore the merit of the message.

I hear what you're saying about workflows for mass-deletes being impacted by this concept, but that doesn't mean there isn't a happy-medium solution. You know... have it as a setting so it can be turned-off (and should be on by default IMO) for those who want it off.

I don't know anything about the FileManager plugin efforts, but if the intent is for OctoPrint to generally continue to have built-in file capabilities, even after this FileManager plugin thing comes into existence, then I stress that it _must_ have an "are you sure?" dialogue capability too (built-in). Because this is software written for humans, and humans make basic mistakes like that.

foosel, 2024-05-29:
You vented your frustration with a ton of caps and punctuation, and complained about the timeline of this issue. You could have said "I would like a confirmation as well" or even made suggestions on how to solve the workflow problem (yes, indeed, stuff like this is also contributing). Instead, you basically screamed into the void, which wasn't a void however but the maintainer. 

That is anything but respectful, or constructive for that matter. You could also now have just said that you are sorry for coming over wrong, or just not said anything. But instead you doubled down with a wall of text on your behaviour being totally friendly and fine and in no way bad, and it's obviously me who needs to grow a thicker skin.

Ok. That's your point of view, your decision to ignore boundaries and your approach to communication. Perfectly fine, but that means you are no longer welcome here.

gabriele-cannizzaro, 2024-08-07:
Hi all, I hope the above hasn't soured the topic too much... I would like to add my thumbs up to this feature request, I think it would be very useful especially in the absence of a "restore" feature, however I appreciate it might be more complex than it looks (references to workflow).

About the reference to a FileManager plugin, are we talking about [this one](https://plugins.octoprint.org/plugins/filemanager/)? I ask because it looks like the last release (and PR) was in Feb 2021, so it looks like it's no longer maintained, but maybe there's discussion somewhere else?

Is there any info on what happens to deleted files? I tried looking for it but couldn't find much. I think that, in the absence of a confirmation dialog or a UI way to list and restore deleted files, having some documentation on how to do it (even if via SSH etc) could help.

Thank you, and thanks for the great work on OctoPrint!

foosel, 2024-09-19:
@gabriele-cannizzaro that's the plugin I was talking about, and I'm happy to report that if all goes according to plan, OctoPrint 1.11.0 will ship with a bundled Upload Manager derived from that plugin, featuring bulk management (copy, move, remove). See #5062 

Consequently, this feature request has been implemented in that PR as well, as the inclusion of the upload manager solves the existing workflow issue (bulk deletion will only require one confirmation instead of one for each deleted item).

I've thus marked it as done.

gabriele-cannizzaro, 2024-09-22:
Hi @foosel, that's great news! One question, will the new file manager be able to restore files deleted before its implementation? I'm not sure I understand whether currently the files get hard-deleted or just moved to a bin location...

Thank you!

foosel, 2024-09-23:
> One question, will the new file manager be able to restore files deleted before its implementation?

No. Deleting something means it's gone. There's no trash can overlay or something like that in OctoPrint or the underlying OS, and that's also not something I'm going to add.

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

commit 668e3f32 (2024-09-19): ux: add a confirmation dialog for file deletion

Now that we have an upload manager that supports bulk
deletion, a confirmation dialog no longer will have as big
of an impact on people's clean-up workflows.

Closes #833
commit c8e5316b (2024-09-23): ux: add a confirmation dialog for file deletion

Now that we have an upload manager that supports bulk
deletion, a confirmation dialog no longer will have as big
of an impact on people's clean-up workflows.

Closes #833
closed 2025-04-22 by foosel