OctoPrint/OctoPrint #834: Upload to SD doesn't.  Raises and ignores an exception.

markwal, 2015-04-03:
1. What were you doing?
Steps:
Connect to printer
Upload to SD
choose: 20mmx1mmcoin.gcode

2. What did you expect to happen?
Attempt to upload indicated file to printer for storage on the SD card

3. What happened instead?
Following exception backtrace, which gets swallowed later

4. Branch & Commit or Version of OctoPrint:
devel 0ed8afd999004d34a0575b11fed38961e4dd4c52

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
I hit it using my printer, but also reproed using VIRTUAL

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Google Chrome 41.0.2272.118 m running on Windows 8.1

7. Link to octoprint.log on gist.github.com or pastebin.com:
https://gist.github.com/c4e0209ae2d01d282f80.git

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

markwal, 2015-04-03:
I think the important bit is in the backtrace in the log:

```
  File "/home/pi/OctoPrint/src/octoprint/printer/standard.py", line 512, in add_sd_file
    remoteName = util.get_dos_filename(filename, existing_filenames=existingSdFiles, extension="gco")
  File "/home/pi/OctoPrint/src/octoprint/util/__init__.py", line 259, in get_dos_filename
    return find_collision_free_name(filename, extension, existing_filenames, **kwargs)
  File "/home/pi/OctoPrint/src/octoprint/util/__init__.py", line 313, in find_collision_free_name
    extension = make_valid(extension)
  File "/home/pi/OctoPrint/src/octoprint/util/__init__.py", line 310, in make_valid
    return re.sub(r"\s+", "_", text.translate({ord(i):None for i in ".\"/\\[]:;=,"})).lower()
TypeError: expected a character buffer object
```

Probably that call to get_dos_filename is supposed to be extension=u"gco"
or make_valid should include a text.decode('utf-8') or that translate map should be the byte array style (not a dictionary).

My guess is the first of those:
```
--- a/src/octoprint/printer/standard.py
+++ b/src/octoprint/printer/standard.py
@@ -509,7 +509,7 @@ class Printer(PrinterInterface, comm.MachineComPrintCallback):
                self.refresh_sd_files(blocking=True)
                existingSdFiles = map(lambda x: x[0], self._comm.getSdFiles())

-               remoteName = util.get_dos_filename(filename, existing_filenames=existingSdFiles, extension="gco")
+               remoteName = util.get_dos_filename(filename, existing_filenames=existingSdFiles, extension=u"gco")
                self._timeEstimationData = TimeEstimationHelper()
                self._comm.startFileTransfer(absolutePath, filename, "/" + remoteName)
```

markwal, 2015-04-05:
Forgive me for adding another bug to this one, but you'll hit it when you're trying out the fix for the above.

I suspect this one won't repro on Windows, just Linux because it's a case sensitivity in file paths bug.

Steps:
1. Connect to the virtual printer with the default config (virtual sd path: ~/.octoprint/virtualSd)
2. Attempt to upload a gcode file

Result
fails because ~/.octoprint/virtualsd doesn't exist (unless it does, in which case it'll fail to show up in the list because it's enumerating the folder virtualSd instead)
```
Proposed patch:
--- a/src/octoprint/util/virtual.py
+++ b/src/octoprint/util/virtual.py
@@ -313,7 +313,8 @@ class VirtualPrinter():
        def _selectSdFile(self, filename):
                if filename.startswith("/"):
                        filename = filename[1:]
-               file = os.path.join(self._virtualSd, filename).lower()
+               filename = filename.lower()
+               file = os.path.join(self._virtualSd, filename)
                if not os.path.exists(file) or not os.path.isfile(file):
                        self.outgoing.put("open failed, File: %s." % filename)
                else:
@@ -500,7 +501,8 @@ class VirtualPrinter():
        def _writeSdFile(self, filename):
                if filename.startswith("/"):
                        filename = filename[1:]
-               file = os.path.join(self._virtualSd, filename).lower()
+               filename = filename.lower()
+               file = os.path.join(self._virtualSd, filename)
                if os.path.exists(file):
                        if os.path.isfile(file):
                                os.remove(file)
```

There also seems to be something wrong with resend that I haven't figured out yet.

jneilliii, 2015-04-11:
I can confirm this error, it's happening for me as well in most recent development version 1.2.0-dev-705-ga34cbc5.

foosel, 2015-04-13:
I just pushed a commit that should fix this.

commit 594636e0 (2015-04-13): Make sure to operate on unicode when sanitzing sd filenames

Should fix #834
closed 2015-04-13 by foosel