OctoPrint/OctoPrint #835: [Bug] -devel- Temperature graph not updated when printer is idle

scp38, 2015-04-03:
1. What were you doing?
Started Octoprint on Raspi, opened GUI in browser. Printer connected, but in idle mode.

2. What did you expect to happen?
Seeing temperature graph being updated every few seconds as well as seeing the temperature commands in the terminal log showing up accordingly, like this
> Recv: T:18.76 /0 B:18.14 /0 B@:0 @:0 T0:18.76 /0 @0:0 T1:18.45 /0 @1:0
> Recv: wait
> Send: N24 M105*17
> Recv: ok 24
> Recv: T:18.76 /0 B:18.14 /0 B@:0 @:0 T0:18.76 /0 @0:0 T1:18.45 /0 @1:0
> Recv: wait
> Send: N25 M105*16
> Recv: ok 25
> Recv: T:18.76 /0 B:18.14 /0 B@:0 @:0 T0:18.76 /0 @0:0 T1:18.45 /0 @1:0

3. What happened instead?
The temperature graph is not updated while the printer is in idle mode and the terminal shows only "Recv. wait"
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait
> Recv: wait

(Note: An update of the temperature is only visible while printing.)

4. Branch & Commit or Version of OctoPrint:
**Devel** branch
**was ok** __up to and including__ commit 3820f7b2752a33fe93b789e1359568860ff97bf6
**broken** __since__ commit 8da9f55b0536a62467892779d9427dd49884e2fe
both commits from March 2nd

5. Printer model & used firmware incl. version
Felix 3.0 duo, Repetier 0.92 

6. Browser and Version of Browser, Operating 
Win7, Firefox 36.0.4

7. Link to octoprint.log on gist.github.com or pastebin.com:
n/a

8. Link to contents of terminal tab or serial.log on 
 n/a

9. Link to contents of Javascript console in the browser 
n/a

10. Screenshot(s) showing the problem 
n/a

I have read the FAQ.

ignaworn, 2015-04-23:
I am also experiencing this issue. Tested today in c6fdab554b258345a506712f690509d4b523e162 and it's still broken.

RUMBA with Repetier Firmware 0.92.3 and Raspberry Pi B+

foosel, 2015-04-23:
Any  ideas why the firmware keeps sending wait responses? Because that strikes me as highly unusual. Usually you should see an M105 that gets then acknowledged with an ok and the current temperatures (normally on the same line per [protocol spec](http://reprap.org/wiki/Gcode#M105:_Get_Extruder_Temperature) but I think OctoPrint's should be lenient enough here to cope). If the firmware instead only keeps sending wait responses, something's off on that end.

foosel, 2015-04-23:
Ok, looks like repetier keeps sending wait to indicate it's buffer is empty. Could any of you please provide a full serial.log from initial connection onward that shows the problem while idle? The except above is not enough to narrow down the issue further.

ignaworn, 2015-04-23:
I don't think it's a firmware issue. I have tested the same setup but instead to a pc with repetier host. M105 polling works fine and you also get those waiting commands (AFAIK is a repetier feature to tell the host that the printer is idle).

Moreover, the printer IS responsive with OctoPrint, moving X, Y etc works, sending a M105 from terminal also works and you get the temperature back as expected.

Here is an actual log from the printer. 
>Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x109bdf0, open=True>(port='/dev/ttyACM0', baudrate=230400, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N1 M105*38
Recv: Info:External Reset
Recv: Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
Recv: Free RAM:3347
Recv: FlowMultiply:100
Recv: ok 1
Changing monitoring state from 'Connecting' to 'Operational'
Send: N2 M105*37
Recv: T:0.00 /0 B:0.00 /0 B@:0 @:0
Recv: ok 2
Send: N3 M21*19
Recv: T:0.00 /0 B:0.00 /0 B@:0 @:0
Recv: ok 3
Send: N4 M20*21
Recv: Unknown command:N3 M21
Recv: ok 4
Recv: Unknown command:N4 M20
Send: N5 M105*34
Recv: ok 5
Recv: T:101.64 /0 B:50.39 /0 B@:0 @:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait

And this is my stripped config.yaml (for connection timeouts) 
>cura: {}
feature:
  alwaysSendChecksum: true
  repetierTargetTemp: true
  sdAlwaysAvailable: true
  sdSupport: false
  waitForStartOnConnect: true
gcodeViewer: {}
plugins:
  discovery:
    httpPassword: null
    httpUsername: null
    model:
      description: null
      name: null
      number: null
      serial: null
      url: null
      vendor: null
      vendorUrl: null
    pathPrefix: null
    publicHost: null
    publicPort: null
    upnpUuid: 00e6d0fb-8399-41c1-ac16-150f493f8aa9
    zeroConf: []
printerParameters: {}
printerProfiles:
  defaultProfile:
    axes:
      e:
        inverted: false
        speed: 300
      x:
        inverted: false
        speed: 6000
      y:
        inverted: true
        speed: 6000
      z:
        inverted: false
        speed: 400
    color: default
    extruder:
      count: 1
      nozzleDiameter: 0.35
      offsets:
      - - 0.0
        - 0.0
    heatedBed: true
    id: _default
    model: Prusa Mendel90
    name: Mendel90
    volume:
      depth: 200.0
      formFactor: rectangular
      height: 200.0
      origin: lowerleft
      width: 200.0
serial:
  autoconnect: true
  baudrate: 230400
  port: /dev/ttyACM0
  timeout:
    communication: 5.0
    detection: 10.0
    temperature: 1.0

foosel, 2015-04-23:
That log makes it look like it initially does some polling iterations
before it then stalls out of the blue. Debug logging (
http://docs.octoprint.org/en/devel/configuration/logging_yaml.html#changing-log-levels)
for octoprint.util.comm and octoprint.util.CountedEvent might be
interesting to figure out where when and why it decides it can't poll any
more.

Btw, it is a firmware issue, but we might be able to find a workaround (it
wouldn't be the first). Repetier is again not sticking to the protocol
here, and that is probably what is causing the issue since it behaves
differently than what octoprint expects (that's what a protocol is for).
I'm not surprised it's working fine with Repetier host, that probably knows
the oddities to expect from its own firmware counterpart.

Note: I'm on CEST, hence no more replies until tomorrow, it's time for some
sleep.

ignaworn, 2015-04-24:
Ok, after almost an hour trying to figure how to enable debugging via ```logging.yaml``` I gave up and started executing ```~/OctoPrint/./run --debug```

BTW, as @scp38 said in the very first, I remember that some time ago, I haven't had any trouble (I haven't upgraded the firmware since then). 
>Branch & Commit or Version of OctoPrint:
Devel branch
was ok up to and including commit 3820f7b
broken since commit 8da9f55
both commits from March 2nd

Tomorrow I'll pull 8da9f55b0536a62467892779d9427dd49884e2fe to see what happens.

Anyway, here's the log. I barely understand what everything means, but is this ok?```octoprint.util.comm - DEBUG - Externally triggered heatup detected```. The printer is idle, and no heaters are enabled.
>2015-04-23 23:41:27,595 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 0
2015-04-23 23:41:27,601 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:28,138 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: -1
2015-04-23 23:41:28,145 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:28,649 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:28,657 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:28,665 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 0
2015-04-23 23:41:28,678 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:28,727 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:28,734 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:28,753 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 0
2015-04-23 23:41:28,762 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:29,155 - octoprint.events - DEBUG - Firing event: Connected (Payload: {'baudrate': 230400, 'port': u'/dev/ttyACM0'})
2015-04-23 23:41:29,168 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x2463ab0>>
2015-04-23 23:41:29,180 - octoprint.util.comm - DEBUG - Externally triggered heatup detected
2015-04-23 23:41:29,198 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:29,205 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:29,214 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 0
2015-04-23 23:41:29,226 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:29,236 - octoprint.util.comm - DEBUG - Externally triggered heatup detected
2015-04-23 23:41:29,248 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:29,256 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:29,263 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 0
2015-04-23 23:41:29,276 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Cleared event
2015-04-23 23:41:29,287 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:29,294 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:29,767 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 2
2015-04-23 23:41:29,775 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:29,782 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - New counter value: 1
2015-04-23 23:41:29,798 - octoprint.util.CountedEvent.comm.clear_to_send - DEBUG - Set event
2015-04-23 23:41:29,804 - octoprint.util.comm - DEBUG - Externally triggered heatup detected

ignaworn, 2015-04-24:
Just to be sure, I recompiled repetier without ```wait``` function
It doesn't work either.
[image]

foosel, 2015-04-24:
The "external heat up detected" is the clue and the acknowledgement order is the key to that issue. The ok for the M105 comes before the answer, so op doesn't know anymore that it requested a temperature so it assumes the firmware is heating up (and hence reporting temps on its own) and then just wants to wait for that to finish. That was added to calculate out the heat up times for better print time estimation and works just perfect with firmware that report the temperature with the ok (like described in the RepRap wiki and like basically every other firmware does it) but doesn't work here. Workaround needed and incoming.

[image]

foosel, 2015-04-24:
And fixed.

foosel, 2015-04-24:
Oh, I should maybe mention that you need to tick this checkbox:

[image]

That will make it work. I added a new feature to the VIRTUAL printer to test and could reliably reproduce the exact behaviour.

closed 2015-04-24 by foosel

ignaworn, 2015-04-24:
Tested today and it's working perfectly, thank you !

scp38, 2015-04-24:
The patch works for me, too, after activating the new checkbox. However, I have the feeling that the temperature updates were more frequent before the issue: only one "wait" in between temperature commands (see my original post), now there are four "wait"s. 

Anyway, back in business again; thanks.

foosel, 2015-04-27:
Before OctoPrint sent an M105 everytime the comm timeout was reached, now it has a proper poll that when idle sticks to the configured polling interval. So if you have a temperature polling interval of more than one second configured, and that `wait` comes once per second, you'll see a couple of waits for each poll.

loloist, 2016-04-17:
Hi. I have a very similar problem. in the log it keeps returning wait over and over again. The issue is that it doesnt poll the printer with m105's in the log, thus i get no temp feedback. The graph  isnt working eighter. ive tried to check the "disable detection of external heatup" in the settings. NO GO. im using the newest stable version of repetier firmware and the newest stable octopi build. how do i get it to work :)         and btw ive added foosel as a user at my octopi. your pass is "foos". non admin rights
IP:46.249.245.56

nbelakovski, 2016-07-25:
I've been having this issue with my Rostock Max V2. I downgraded OctoPrint to 3820f7b as mentioned in the original post and it's working fine now. I'd like to make some time to go through this thread more thoroughly and maybe dive into the issue and come up with a better solution, but I'm just leaving this comment for now so that anyone else with a Rostock might be better able to find this thread in searches :)

foosel, 2016-07-25:
[From the contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report):

> ### What should I include in a bug report?
> 
> Always use the following template (you can remove what's within [...], that's only provided here as 
> some additional information for you), **even if only adding a "me too" to an existing ticket**:

Please don't make me fly blind :/

nbelakovski, 2016-07-25:
My apologies! I'm new to contributing! I will try to find the time sometime this week, or more likely this weekend, to follow the template. For now I can tell you that it was pretty much identical to the initial report, with the exception of the printer involved, and I'm mostly running from Chrome on various Linux/Mac machines. Will try to be more diligent upfront next time!

augman6, 2017-05-29:
http://imgur.com/gallery/2eNg5
I am in the master branch 1.3.2 and the setting "disable detection of external heatup" completely. Did you get rid of it in an update by mistake or was this joined with something else, or am I missing somethings all together?

foosel, 2017-05-29:
@augman6 not removed, but moved into the set of features covered by automatic firmware detection. When you disable that, you can still manually configure the firmware relevant settings:

[image]

commit 9ee9f7bf (2015-04-24): Workaround for idle temperature polling not working with Repetier Firmware

Repetier always first sends the ok and then any command output. In case of M105, that makes the response look like from an externally triggered heatup (no ok on the same line), causing polling to stop until that falsely detected heatup is complete. Added a configuration option to disable heatup detection. The disadvantage of this is that when printing via Repetier Firmware from SD, the heatup times won't be substractable from the total print time, leading to a less accurate print time left estimation.

 Closes #835