OctoPrint/OctoPrint #837: [Brainstorming] High resolution support for gcode viewer

lucas-clemente, 2015-04-03:
This isn't a high priority, but the gcode viewer currently doesn't seem to support high-resolution (retina) screens.

I will try to look into it over the next few days and see whether it's easy to implement (e.g. using window.devicePixelRatio to scale the canvas).

To please your bot (and since I wan't allowed to reopen #836):

What were you doing?
Using octoprint.

What did you expect to happen?
Retina gcode viewer.

What happened instead?
Non-retina gcode viewer.

Branch & Commit or Version of OctoPrint:

Printer model & used firmware incl. version
(if applicable - always include if unsure):

Browser and Version of Browser, Operating 
System running Browser (if applicable - always 
include if unsure):

Link to octoprint.log on gist.github.com or pastebin.com:

Link to contents of terminal tab or serial.log on 
gist.github.com or pastebin.com (if applicable - always 
include if unsure or reporting communication issues):

Link to contents of Javascript console in the browser 
on gist.github.com or pastebin.com or alternatively a
screenshot (if applicable - always include if unsure
or reporting UI issues):

Screenshot(s) showing the problem (if applicable - always
include if unsure or reporting UI issues):

I have read the FAQ.

foosel, 2015-04-13:
The highest resolution I have available for testing here is Full HD, so I can't help here I fear other than by hinting at where to look for stuff.

foosel, 2015-05-05:
Fixed by your own PR, thank you! :)

PR #858 (2015-05-05): Add support for high-DPI screens to gcode renderer
Fixes #837.

I've tested it in Safari and Chrome on OS X, would be cool if somebody with access to more browsers could have a look.

Old:

[image]

New:

[image]
closed 2015-05-05 by foosel