OctoPrint/OctoPrint #841: [Bug] [devel] Can't login using IE

markwal, 2015-04-07:
1. What were you doing?
Browse to OctoPrint
Click on login
Click on the input box

2. What did you expect to happen?
To be able to type username and password

3. What happened instead?
Popup disappears

4. Branch & Commit or Version of OctoPrint:
devel a34cbc5

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Internet Explorer 11.0.9600.17690 on Windows 8.1

7. Link to octoprint.log on gist.github.com or pastebin.com:
https://gist.github.com/markwal/7ff6d6ce64246f9b986c

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

I think this is because main.js bails out when creating the view models because it doesn't have a name property on function objects.  And since the rest of the module doesn't run, we don't get the even handler that stopPropagation on the input box.  This patch seems to fix it:

```
diff --git a/src/octoprint/static/js/app/main.js b/src/octoprint/static/js/app/main.js
index ed33b1c..cbf6950 100644
--- a/src/octoprint/static/js/app/main.js
+++ b/src/octoprint/static/js/app/main.js
@@ -57,6 +57,16 @@ $(function() {
         // the view model map is our basic look up table for dependencies that may be injected into other view models
         var viewModelMap = {};
 
+        // Fix Function#name on browsers that do not support it (IE):
+        // see: http://stackoverflow.com/questions/6903762/function-name-not-supported-in-ie 
+        if (!(function f() {}).name) {
+            Object.defineProperty(Function.prototype, 'name', {
+                get: function() {
+                    return this.toString().match(/^\s*function\s*(\S*)\s*\(/)[1];
+                }
+            });
+        }
+
         // helper to create a view model instance with injected constructor parameters from the view model map
         var _createViewModelInstance = function(viewModel, viewModelMap){
             var viewModelClass = viewModel[0];
```

foosel, 2015-04-13:
Cherry picked your fixing commit on `devel`, big thanks!

closed 2015-04-13 by foosel

gorbehnare, 2015-07-03:
Hello

I have the exact same issue described above on any browser from any computer (so this is not a browser issue). the issue is as soon as I click on the login username and password fields they disappear. Even if I succeed in filling up the fields by using Tab the login button does not work. Did the code issues made it's way to Version: 1.1.1-30-g4fede5a (master branch)?

foosel, 2015-07-03:
No, please update.

gorbehnare, 2015-07-03:
I have updated to Version: 1.2.2 (master branch), rebooted the entire thing, and still the same issue.

foosel, 2015-07-03:
Can you please stop posting to two tickets concurrently? It makes it quite hard. Please also make sure you have refreshed your browser's cache (as state in the other ticket you posted too). If that doesn't help, open a new ticket. Follow *everything* in https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report

gorbehnare, 2015-07-03:
sorry about that. This seemed relevant to both tickets. I have tested this on 3 computers and 2 or 3 browsers on these computers. I can verify that this is a UI/software issue and not a browser issue. I even tried it on the rpi itself and same issue. I'm about to give up on this and reimage the SD and try again. Not sure if there is log I can pull for an issue like this?

foosel, 2015-07-03:
There is, the javascript error console, also referred to in the linked document. All I can say is is that right now I cannot reproduce it, so we need more information to fix this, and it certainly doesn't belong into this ticket here because that was actually a bug that was restricted to IE and has been fixed and verified. Open a new ticket and we'll take it from there. But I'm fairly sure it is some sort of caching issue or something like it because I just setup up three new instances yesterday from 1.2.2 and had no problems like that at all across multiple browsers.