OctoPrint/OctoPrint #844: Latest devel version is incompatible with flask >=0.10

Athemis, 2015-04-09:
JSONBadRequest has been removed from flask version >=0.10 (http://flask.pocoo.org/docs/0.10/changelog) and leads to octoprint crashing at startup:

1. What were you doing?
Run latest development version of octoprint

2. What did you expect to happen?
It should have started up

3. What happened instead? 
Crashed with the following error

```
2015-04-09 21:00:03,219 - octoprint.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/usr/share/octoprint/run", line 22, in <module>
    octoprint.main()
  File "/usr/share/octoprint/src/octoprint/__init__.py", line 81, in main
    octoprint.run()
  File "/usr/share/octoprint/src/octoprint/server/__init__.py", line 690, in run
    from octoprint.server.api import api
  File "/usr/share/octoprint/src/octoprint/server/api/__init__.py", line 30, in <module>
    from . import printer as api_printer
  File "/usr/share/octoprint/src/octoprint/server/api/printer.py", line 9, in <module>
    from flask.exceptions import JSONBadRequest
ImportError: No module named exceptions
```

4. Branch & Commit or Version of OctoPrint:
devel@a34cbc58dc2460a848b3244690c842253d0d3681

5. Printer model & used firmware incl. version
Kossel Mini & Marlin 1.0.2

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/41.0.2272.118 Safari/537.36

7. Link to octoprint.log on gist.github.com or pastebin.com:
None saved

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

foosel, 2015-04-13:
The requirements are pinning Flask to 0.9 (compare [this line](https://github.com/foosel/OctoPrint/blob/devel/setup.py#L14)), how did you stumble across this, did you manually upgrade Flask?

Athemis, 2015-04-14:
Partly. I already had flask installed using the system package manager. On ArchLinux the flask version in the repo is 0.10. It didn't come to my mind to check the setup.py file, because the latest stable version of octoprint works just fine with flask-0.10.
Obviously I'll have to install it using virtualenv pulling in all dependencies manually.

foosel, 2015-04-14:
While that is a good idea in general, I'll nevertheless see that I get this compatibility issue resolved.

foosel, 2015-04-22:
@DanLipsitt was faster than me, see #865

closed 2015-04-22 by foosel