OctoPrint/OctoPrint #859: Printing Slower than Set Speeds

fdmprocess, 2015-04-18:
I'm running raspian on my solidoodle 3 with an rPi B+, everything seems to be working fine except for my printer is running slower than the speeds I set in rHost. Does anyone have experience with this?

closed 2015-04-18 by GitIssueBot

nophead, 2015-04-18:
Print speed is determined by the slicer and the firmware, not the host.

fdmprocess, 2015-04-19:
No shit. I used the slicer in rHost (repetierHost) and my print speed is running slower than what I set in the slicer.

nophead, 2015-04-19:
OctoPrint just sends the gcode to the printer. It doesn't modify the F values in the code to change the speed. Have a look at the gcode and see if the F values match what you expect.

fdmprocess, 2015-04-19:
I think the real culprit is that I am trying to stream too high of a resolution from my webcam and that is eating up too many resources. I'll report back after some fooling around. Sorry about the short response earlier but someone who us attempting to use octoprint with their printer probably knows where their gcode is coming from.

nophead, 2015-04-19:
When that happens the moves will be the right speed but have pauses between them so you get blobs. If the object has lots of small segments and no long ones the effect is that it all slows down as the firmware can't blend the moves and so always starts each one from a low speed.

jneilliii, 2015-04-19:
Yeah, I was having this issue with certain prints that were more complicated. As recommended to me on the Google group page, printing from sd card on my printer board and not the pi worked like a charm.

jlipavsky, 2018-01-15:
that makes the raspberry pi useless then...... and yet no one knows the problem... printing from an sd isnt a solution ......