OctoPrint/OctoPrint #861: Print button not working (sometimes)

quillford, 2015-04-18:
1. What were you doing? Trying to start a print with the Print button in the 'State' panel

2. What did you expect to happen? The Print to start

3. What happened instead? Nothing

4. Branch & Commit or Version of OctoPrint: Version: 1.2.0-dev-729-gc5484fe (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure): Kossel 2020 with Smoothieware

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure): Chrome 42.0.2311.90

7. Link to octoprint.log on gist.github.com or pastebin.com: http://www.hastebin.com/zaporidulu.vhdl

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues): 

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues): POST http://kossel.local/api/job 500 (INTERNAL SERVER ERROR)

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

[image]
[image]

Sometimes, the print button in the 'State' panel does not work. What I have to do is select another file and then press the print button on the file. It seems the {command: "start"} call doesn't always work, but {command: "select", print: true} works.

I have read the FAQ.

foosel, 2015-04-22:
Thanks for reporting. Are you maybe reconnecting after selecting a file? So something like:

1. Connect
2. Select
3. Disconnect
4. Connect
5. Print (file still shown as selected) => **Boom**

If so, that's a known bug, but it doesn't have a ticket yet (it only exists on my whiteboard so far), really have to tackle that one.

foosel, 2015-04-22:
@quillford IF your problem is the same as the one I observe, the above patch should fix it (by deselecting the file if the printer is disconnected - keeping it selected is too risky internally, could easily cause even more hard to track down synchronization issues, so that's the clean way). Can you verify?

closed 2015-04-24 by quillford