OctoPrint/OctoPrint #863: [Request] Add IFTTT Support

Mauronic, 2015-04-19:
A lot of easy to setup and use features could be added with IFTTT integration. For example:

- Turn off my printer when finished printing (using IoT outlet / hub) <-- my favorite
- Email me when my print is done
- Send me a picture of the object when it is done or post it to social media
etc

I know you could do some of this now with some programming but the IFTTT UI is very handy.

markwal, 2015-04-19:
You might want to check out: https://github.com/foosel/OctoPrint/wiki/Cookbook:-Useful-Eventhooks some of the things you mention are possible with a little configuration.  The page also mentions using the e-mail hook to send a mail that ifttt can read.  You could use that to drive other rules on ifttt.

foosel, 2015-04-19:
The problem here is that ifttt doesn't have an API that could be used by third party software, certainly not by individually installed open source software. They curate their partners quite heavily. So "workarounds" like triggering mails or doing stuff like adding lines to google drive documents or posts to WordPress blogs are the only option to get some kind of integration working here.

icecube45, 2015-06-18:
[With the announcement of the maker channel] (http://blog.ifttt.com/post/121786069098/introducing-the-maker-channel) , implimentation into IFTTT might be possible.

foosel, 2015-06-18:
Very good news, thank you for the heads up!

foosel, 2015-06-18:
On first look somewhat limited though. No additional data on events you can send them, just an event name and that's it. Well, but at least more than was possible previously.

icecube45, 2015-06-18:
To be honest, I didn't look to far into it (heh).
Glad you can at least send 3 variables.

markwal, 2015-06-20:
Ok, I did the dumbest possible plugin for the Maker channel at https://github.com/markwal/OctoPrint-Pfttt just for experimentation.  It just spams every event to the channel.  We probably want to be more selective.  But as it stands, it can do the first two of OPs list.

Here's some things I've learned so far:
- Since the channel leaves the event name open ended it makes wiring it up in IFTTT not for the average user since there isn't any pick list.  You have to type the event name precisely.
- Those value(s) can detect the types of the incoming objects somewhat (like it'll turn a http link into a short link through ifttt.com), but it doesn't like nested json very much and I'm not sure there's a way to send anything like a photo through it.

To accomplish that third thing, it seems like we'd still need a plug in to upload the picture automatically somewhere and the ifttt event would have the link. But that jumps ahead and solves the problem of sending the pic without ifttt.  Or you make the picture retrievable directly off the pi (via punching holes through your firewall) which isn't any easier to set up that coding directly the upload to flickr or dropbox, so I'm not sure ifttt helps.  Though I could make a plugin that has a direct wiring for dropbox, onedrive or gdrive and/or YouTube and also sends an event to IFTTT so perhaps that opens up some further action.  Though with the auto upload you can already trigger off of the upload event on the other channel (like dropbox channel).  So... anyway.

chasecromwell, 2016-01-24:
I plan on working on this some more, just to give a heads up.

chasecromwell, 2016-01-24:
I have set up the JSON part for IFFT but I'm not sure about which events to trigger for.
Do these sound right?
Print Finished
Print Canceled
Print Started
File Uploaded
File Deleted
Temperature Met.

All of these would be turnable on and off in octoprint and I'm setting up 2 for now, one for a notification, and the other for a SMS to a configurable number.

markwal, 2016-01-24:
@chasecromwell That plugin at OctoPrint-Pfttt has an example of using "on_event" BTW.

chasecromwell, 2016-01-24:
Ah, thank you @markwal. I'm forking your repo and should have a basic configurable plugin some time toady.

foosel, 2017-01-27:
Closing this since there is a plugin and that's also about as good as it will get - IFTTT is too closed otherwise, no way to integrate with them in another way.

closed 2017-01-27 by foosel

ieatacid, 2018-05-07:
I know this is old, but IFTTT added [maker webhooks](https://ifttt.com/maker_webhooks) so you can send a notification to your device of choice using OctoPrint's events.

1. Go to [https://ifttt.com/maker_webhooks](https://ifttt.com/maker_webhooks) and click the big Connect button
2. On the next page click [Settings](https://ifttt.com/services/maker_webhooks/settings)
3. Below 'Account Info' take note of your key at the end of your URL (after /use/): `https://maker.ifttt.com/use/h4tH5Xx3OsI1l_WfkaP8ARec4j-vrJm3Uk0tzlAeUDS`
4. Paste the following into your config.yaml, replacing the string of 'x' characters with your key:
```yaml
events:
  enabled: True
  subscriptions:
  - event: PrintDone
    command: curl -s
      -F "value1={name}"
      https://maker.ifttt.com/trigger/print_done/with/key/xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    type: system
```
5. Save, restart OctoPrint
6. In the IFTTT app create a new applet and in the Search field for selecting a trigger service type 'hook'. The only result will be 'Webhooks', click that. Click 'Receive a web request' on the next screen.
7. On the next screen, 'Complete trigger', type in 'print_done' (without quotes) for the Event Name, then click Continue
8. Click '+That' and choose 'Notifications' (or whatever you like) and then click 'Send a notification from the IFTTT app
9. For the message I erased what was there, type in 'Print done', clicked 'Add ingredient' then 'Value1' (Value1 will give you the file name that just printed). When done click 'Create action'.
10. Click 'Finish' and you're done!
<br>
I could only get it to work using that exact syntax above in config.yaml. Any other way I tried it would either crash OctoPrint with a CRITICAL octoprint.server error or it just wouldn't start at all and nothing would show in the log file.

<br><br>

@foosel - maybe you can update the [event cookbook](https://github.com/foosel/OctoPrint/wiki/Cookbook:-Useful-Eventhooks)?

foosel, 2018-05-08:
@ieatacid it's a wiki, feel free to update it yourself ;)