OctoPrint/OctoPrint #869: [Support] Connection

marvinstuart, 2015-04-23:
Using latest  Version: 1.2.0-dev-783-gc6fdab5 (devel branch)

Issue:
When i click on connect in says connecting and within 20 secs in stops connecting and says closed. This issue only started with the update last week.

Terminal information:
Recv: start
Recv: echo: External Reset
Recv: Marlin1.0.2
Recv: echo: Last Updated: Apr  2 2015 11:39:59 | Author: (none, default config)
Recv: Compiled: Apr  2 2015
Recv: echo: Free Memory: 5125  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z1600.00 E105.67
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X800 Y700 Z100 E10000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S700.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P12.70 I0.51 D79.27
Recv: echo:Filament settings: Disabled
Recv: echo:Z-Probe Offset (mm):
Recv: echo:   M851 Z-1.10
Changing monitoring state from 'Connecting' to 'Closed'
Connection closed, closing down monitor

foosel, 2015-04-23:
Next time please stick to the contribution guidelines and the bug report template, thanks.

Does increasing your communication timeout do anything? Because that just does look like your printer taking too long to come fully online after OctoPrint connects to it, which might have been stopped from happening earlier due to slightly different implementation details.

marvinstuart, 2015-04-23:
Changed communication timeout to 90s and changed connection time to 30s and same issue see below

Recv: start
Recv: echo: External Reset
Recv: Marlin1.0.2
Recv: echo: Last Updated: Apr  2 2015 11:39:59 | Author: (none, default config)
Recv: Compiled: Apr  2 2015
Recv: echo: Free Memory: 5125  PlannerBufferBytes: 1232
Recv: echo:Hardcoded Default Settings Loaded
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z1600.00 E105.67
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X800 Y700 Z100 E10000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S700.00 T1000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P12.70 I0.51 D79.27
Recv: echo:Filament settings: Disabled
Recv: echo:Z-Probe Offset (mm):
Recv: echo:   M851 Z-1.10
Changing monitoring state from 'Connecting' to 'Closed'
Connection closed, closing down monitor

foosel, 2015-04-23:
Please provide a screenshot of your Settings > Features page (I want to see which options you have checked there), a full serial.log that also includes the part below (I want to see what connection options are used for creating the serial connection) and what printer and printer electronics that is (I see that the firmware is Marlin).

```
Connecting to: COM12
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x60b7c30, open=True>(port='COM12', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
```

What would also be great (if possible) would be the last version it worked for you (since I work full time on this, "since last week" is a bit too vague, too many changes in between).

marvinstuart, 2015-04-23:
[image]

Link to Log
http://marvinstuart.com/file/

Sorry i honest dont remember the last working version number

marvinstuart, 2015-04-23:
Issue Fixed i feel stupid and i needed to do is to make sure that  Wait for start on connect was checked

foosel, 2015-04-25:
For another report on the mailing list about this and want to get to the ground of this.

sprior, 2015-04-25:
Looks like turning on "wait for start on connect"

jinschoi, 2015-04-27:
Same here. Without that checked, it looks like it is attempting to send M105 immediately, before the printer is ready for it, then times out waiting for the response. I suspect it has something to do with commit 5804825dc3f085f071e2879a7d675f3be033cbcb.

foosel, 2015-05-04:
Should be fixed in current `devel`, can you verify?

jinschoi, 2015-05-04:
Seems to work. Here is a Terminal trace. It is now attempting to send M110 twice, before Marlin resets, so I'm not sure those are having any effect.

```
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x1235030, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M110
Recv: start
Send: M110
Recv: echo: External Reset
Recv: Marlin 1.0.0
Recv: echo: Last Updated: Aug  6 2014 00:56:28 | Author: (none, default config)
Recv: Compiled: Aug  6 2014
Recv: echo: Free Memory: 4141  PlannerBufferBytes: 1232
Recv: echo:Stored settings retrieved
Recv: echo:Steps per unit:
Recv: echo:  M92 X88.88 Y88.88 Z400.00 E471.50
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z25.00 E25.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E10000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S2000.00 T3000.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P16.15 I0.93 D70.30
Recv: echo:SD init fail
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: M105
Recv: ok T:18.3 /0.0 B:20.6 /0.0 T0:18.3 /0.0 T1:18.7 /0.0 @:0 B@:0
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M105
Recv: ok T:19.6 /0.0 B:20.6 /0.0 T0:19.6 /0.0 T1:18.7 /0.0 @:0 B@:0
Send: M105
```

foosel, 2015-05-04:
That looks ok. The first send makes sense for boards that don't reset upon connect (e.g. Printrboards), otherwise you wouldn't get a communication going at all (since the firmware wouldn't send anything by itself), so that's fine.

foosel, 2015-05-04:
Marking this as closed for now, in case some one still runs into this problem please request to reopen.

closed 2015-05-04 by foosel

sprior, 2015-05-05:
Last Fall I installed OctoPrint from the devel branch and had no problem if I printed something, then turned off the printer, then later turned the printer back on - OctoPrint would re-establish the connection and I could warm up the printer and print with no issue.  Earlier this year my pi was corrupted and I reinstalled with the current devel branch and ran into this bug, but also once I turned off my Makergear M2 for a period of time and then turned it back on I was unable to set temperatures until OctoPrint was restarted (usually accomplished by rebooting the pi).  I was hoping the fix to this bug would also help with the restart issue, but tonight I pulled from the updated devel branch, made a test print, powered off the printer for a few minutes, then turned it back on and I was still unable to turn on the hotend or bed.  I went into the terminal window and these were the last few lines:
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Z0.40 E5.00
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P17.92 I1.11 D72.23
Recv: echo:SD init fail

Can other M2 owners see if they also still cannot automatically reconnect after the printer was power cycled?

commit e96dcadc (2015-04-27): Fix: Receiving a "start" from the firmware triggers another handshake attempt

Fixes #869