OctoPrint/OctoPrint #878: Cura Profile Import Error

quillford, 2015-04-28:
1. What were you doing? Importing a Cura profile for slicing within OctoPrint

2. What did you expect to happen? The profile to import

3. What happened instead? Error (400 Bad reqeust)

4. Branch & Commit or Version of OctoPrint: 1.2.0-dev-705-ga34cbc5 (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure): irrelevant

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure): Chrome 42.0.2311.90 (Official Build) (64-bit)

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE): [octoprint.log](https://gist.github.com/quillford/5c10c5ca776be195adc4)

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues): http://hastebin.com/nujoqezewa.hs

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues): [image]

I have read the FAQ.

foosel, 2015-05-04:
Can you please provide the profile with which you are experiencing these problems? Thanks

quillford, 2015-05-07:
http://hastebin.com/xacenimozi.avrasm

foosel, 2015-05-07:
Somehow I'm really lucky these days when it comes to bugs I can't reproduce it seems... Rolled back to your version, tried importing your profile, worked flawlessly :/ I first thought it might be an old bug I thought I'd fixed that caused importing not to work when the path to the CuraEngine binary was not set, but when I removed that I still could import just fine, so it's not that.

I'll try the following: I'll add some more validation logging to the cura plugin. That should then at least give us a hint why your profile can't import. It would be great if you would also update to the current devel so we can be sure the problem still persists there (probably, since if it's happening for you on 705 and not for me, chances are high it's still happening for you on the current HEAD although it's not happening for me, haven't touched the plugin in a while).

foosel, 2015-05-08:
@quillford I added more logging, can you update and report if a) the issue still persist or not and b) if it still persists provide another log (this time it should hopefully include more helpful information after the commit above).

markwal, 2015-05-08:
Perhaps version of CuraEngine is relevant?

foosel, 2015-05-09:
No, CuraEngine isn't involved in the profile management part, that only gets active once you actually want to slice something.

foosel, 2015-05-09:
On closer look, that actually looks like a duplicate of #795. Closing this for now (please correct me if I'm mistaken in assuming this to be the same, symptoms sound identical), but @quillford please provide information as requested above!

closed 2015-05-09 by foosel

quillford, 2015-05-09:
Sorry. The printer I use slicing within OctoPrint is down. I'll try it right now. But that issue seems similar.