OctoPrint/OctoPrint #882: Multiple regex controls

MoonshineSG, 2015-05-03:
When using multiple custom controls with regex, the templates get crossed updated... (and sometimes the multiple regex confuse each other... ) 

Are there restriction on single use of regex controls ?

MoonshineSG, 2015-05-03:
1. What were you doing?
added regex custom controls to config.yaml
 
2. What did you expect to happen?
have different templates for content triggered and matched by regex

3. What happened instead?
the templates get crossed updated... (and sometimes the multiple regex confuse each other... )

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-668-g3ebd54d (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
n/a

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Chrome OSX

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
n/a

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):
n/a

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
n/a

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
n/a

foosel, 2015-05-04:
If the regexes "overlap", that sounds like valid behaviour on first look, but if you would provide an example of your configuration, it might be easier to determine.

MoonshineSG, 2015-05-04:
I see... So the template matching is not done independently for respective regex ?
If thats true, then yes, the one button push update 2 templates would be correct. 

The following (partial) config fails. The first button doesnt work (no updates), but the second button updates both templates (as expected

- command: M114
  name: Get Position
  regex: X:([0-9.]+) Y:([0-9.]+) Z:([-0-9.]+) E:([0-9.]+)
  template: 'Position: X={0}, Y={1}, Z={2}, E={3}'
[... other buttons without regex ...]
  - command: M114
    name: Step 3 (read value)
    regex: X:([0-9.]+) Y:([0-9.]+) Z:([-0-9.]+) E:([0-9.]+)
    template: 'Set Z={2} [X={0}, Y={1}]'

foosel, 2015-05-04:
In fact, all regexes are merged into one for performance reasons. If OctoPrint tried to match any regex against each incoming line, that would be really bad for throughput.

But I did not think of overlapping regexes when implementing this. I'll test with your config there and see if anything can be done about it that doesn't negatively influence performance.

Would probably be an idea to only "enable" the regex if the corresponding command was actually detected, and disable matching for it again once a response was seen. WIll have to think about it.

MoonshineSG, 2015-05-04:
As a workaround I took out my second regex and template and when clicking second button I read the values from the first template (which gets updated correctly and it's only a few rows above)

foosel, 2015-05-04:
Good news, that behaviour has already been fixed in the current `devel` - please try updating, then your multiple regexes should work just fine. Processing was changed since your version so that multiple regexes with different templates are no problem.

closed 2015-05-04 by foosel

MoonshineSG, 2015-05-05:
Yes, it seams to work well. Of course since both my regex match the output of the same command, bothe templates get updated regardless which button gets pressed. But that's fine. Thanks.

Olevinn, 2016-11-09:
Hy,
my version: 1.2.17 (stable)
plugin: Custom Control Editor
I add "command" and i checked use output. My problem: only 1 parameter working.
Working:
`regex: X:([-+]?[0-9.]+)`
`template: X={0}` 
but more parameter dont work.
Not work:
`regex: X:([-+]?[0-9.]+)Y:([-+]?[0-9.]+)` 
`template: X={0},Y={1}` 
or worked version 
template another variation, template: x axis: {0} <= not work

foosel, 2016-11-09:
This:

``regex: X:([-+]?[0-9.]+)Y:([-+]?[0-9.]+)``

won't match a string such as:

``X:123.45 Y:234.56 Z:345.67``

because it's missing the whitespace in between the number after ``X`` and the string ``Y:``.

Try:

``regex: X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+)``

Olevinn, 2016-11-09:
`regex: X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+)`
not work :(

foosel, 2016-11-09:
Is that Marlin output *really* without whitespace between the various coordinates? What kind of printer is that, and what Marlin version? This looks like yet another case of broken firmware, stuff like this is driving me nuts.

If the output indeed is missing the spaces, then your original regex should work.

Since it doesn't, I'll need to take a look at this, but I won't be able to do so today and probably also not tomorrow.

Olevinn, 2016-11-10:
Marlin version: 1.x
Machine: Velleman K8200 (base, custom rebuilt, Zaxiy, extruder, motor control, mechanic)

I know how to get debug information? (Regex function monitoring, and template variables)

MoonshineSG, 2016-11-10:
I tried on mine as well, and the regex seams to have a problem. (it's been a while since i used that custom button and I didn't notice the error)

regex : `X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)`
marlin output: `X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:0`

tested on `https://regex101.com/` and the values are identified correctly. OctoPrint does not output them .

foosel, 2016-11-11:
Found a bug in `devel`/1.3.0rc1 that caused feedback controls to not work, see #1588. Has since been fixed in `devel` and will be rolled out with 1.3.0rc2.

However, on 1.2.17 I cannot reproduce and in fact the above regex works fine here (after modifying the Marlin output to be broken and missing the spaces like in the example provided by @Olevinn).

In short: this config

```
- command: M114
  name: Broken Position
  regex: X:([-+]?[0-9.]+)Y:([-+]?[0-9.]+)
  template: 'Broken Position: X={0}, Y={1}'
```

works on 1.2.17 (and current `devel` but *not* 1.3.0rc1) against this output

```
Send: M114
Recv: ok X:10.00Y:3.20Z:5.20E:1.24 Count: A:1000 B:320 C:1040
```

(which - again - is broken, and will cause problems going down the road - Velleman broke something in their Marlin version here by stripping the spaces and needs to fix that, tell them that, it's their firmware they broke!)

Olevinn, 2016-11-15:
My regex worked, but any change -> reboot system -> working

I can do that after button of the click change the color?

foosel, 2016-11-15:
> My regex worked, but any change -> reboot system -> working

I don't understand, can you rephrase?

> I can do that after button of the click change the color?

No

Olevinn, 2016-11-15:
I anything change the regex, and save config files does not working. Reboot the system and it works (if the correct regex)

Olevinn, 2016-11-15:
Change color: ok, no..
then
how to create progress bar?

foosel, 2016-11-15:
> I anything change the regex, and save config files does not working. Reboot the system and it works (if the correct regex)

Disconnecting from the printer and reconnecting should also suffice I think. But yes, most changes to the `config.yaml` file require a restart of OctoPrint (but not a reboot)

> how to create progress bar?

Also not possible for simple custom controls, you'd need to write a full fledged plugin for that. 

But please seek additional support on the [mailing list](https://groups.google.com/group/octoprint) or in the [G+ community](https://plus.google.com/communities/102771308349328485741), this is a closed bug report that is pretty unrelated to your additional questions, not a support forum :)