OctoPrint/OctoPrint #886: Access Control stuck in loop

quillford, 2015-05-09:
1. What were you doing? Installing OctoPrint on a new machine.

2. What did you expect to happen? The Access Control modal to go away once I pressed disable Access Control

3. What happened instead? It would persist

4. Branch & Commit or Version of OctoPrint: 1.2.0-dev-832-gfa98210 (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure): Chromium 41.0.2272.76

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE): https://gist.github.com/quillford/e3093c293ab30aa4684a

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues): 

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues): http://youtu.be/ZXGm2PjBRiQ

I have read the FAQ.

quillford, 2015-05-09:
Note, the issue does go away upon restarting OctoPrint.

foosel, 2015-05-09:
Issue goes away as well when shift-reloading the page, thus forcing a cache invalidation. It's caused by either the browser or server side cache. Removed the automatically triggered reload after disabling access control that was causing that for now, need to find a way that reliably clears the caches before reintroducing it (it's needed in order to not have the navigation bar show the login information after disabling access control, but I'd rather have that show up there with user "dummy" until a reload than have the dialog be stuck in an infinite loop until a reload)

commit 95688062 (2015-05-09): Fix: Don't reload page when disabling access control in first run dialog

Cache clearing doesn't work reliably with location.reload(true), leading to cache induced infinite loop when disabling access control (since generated template still assumes first run dialog to be active). Disabling for now, need to find a better way to reload the page while clearing the browser's and server's cache (same issue exists for plugins needing a client side triggered reload).

Fixes #886
closed 2015-05-09 by foosel