OctoPrint/OctoPrint #887: `ZeroDivisionError` on startup

jminardi, 2015-05-09:
1. What were you doing?

starting octoprint

2. What did you expect to happen?

no error

3. What happened instead?

ZeroDivisionError

4. Branch & Commit or Version of OctoPrint:

OctoPrint 1.2.0-dev-832-gfa98210

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

rambo

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

safari latest

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

```
~/OctoPrint $ ./run
2015-05-09 21:58:26,842 - octoprint.server - INFO - Starting OctoPrint 1.2.0-dev-832-gfa98210 (HEAD branch)
2015-05-09 21:58:26,846 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/.octoprint/plugins, /home/pi/OctoPrint/src/octoprint/plugins and installed plugin packages...
2015-05-09 21:58:26,967 - octoprint.plugins.discovery - INFO - pybonjour is not installed, Zeroconf Discovery won't be available
2015-05-09 21:58:26,969 - octoprint.plugin.core - INFO - Found 3 plugin(s) providing 3 mixin implementations, 1 hook handlers
2015-05-09 21:58:26,980 - octoprint.plugin.core - INFO - Initialized 3 plugin(s)
2015-05-09 21:58:26,982 - octoprint.plugin.core - INFO - 3 plugin(s) registered with the system:
|  CuraEngine (bundled) = /home/pi/OctoPrint/src/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/OctoPrint/src/octoprint/plugins/discovery
|  Virtual Printer (bundled) = /home/pi/OctoPrint/src/octoprint/plugins/virtual_printer
2015-05-09 21:58:26,983 - octoprint.filemanager - INFO - Adding backlog items from local to analysis queue
2015-05-09 21:58:29,416 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2015-05-09 21:58:29,539 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on raspberrypi for SSDP
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/OctoPrint/src/octoprint/filemanager/analysis.py", line 176, in _work
    self._analyze(entry)
  File "/home/pi/OctoPrint/src/octoprint/filemanager/analysis.py", line 193, in _analyze
    result = self._do_analysis()
  File "/home/pi/OctoPrint/src/octoprint/filemanager/analysis.py", line 241, in _do_analysis
    self._gcode.load(self._current.absolute_path, self._current.printer_profile)
  File "/home/pi/OctoPrint/src/octoprint/util/gcodeInterpreter.py", line 39, in load
    self._load(f, printer_profile)
  File "/home/pi/OctoPrint/src/octoprint/util/gcodeInterpreter.py", line 152, in _load
    totalMoveTimeMinute += math.sqrt(diffX * diffX + diffY * diffY) / feedRateXY
ZeroDivisionError: float division by zero
```

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
NA
10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
NA
I have read the FAQ.

foosel, 2015-05-09:
Uploaded a file recently that contains a gcode line with "F0" in it or set x or y axis speed to 0?

foosel, 2015-05-09:
Shouldn't happen anymore with above commit, however that you ran into that error at all hints at either a brokenly configured printer profile or a gcode file containing garbage feedrates, so better check both (see my last comment for what to look for).

closed 2015-05-09 by foosel

jminardi, 2015-05-10:
Wow, thanks for such a fast fix. And you're right I had a 0 in there somewhere.

commit 5fa30e41 (2015-05-09): Ignore feed rates of 0, those don't make sense anyhow

Fixes #887