OctoPrint/OctoPrint #891: save options

MoonshineSG, 2015-05-13:
What were you doing?
n/a

What did you expect to happen?
Under the Terminal it would be useful if the state of the filters (on/off) could be remembered between session/reboots. Same thing under the GCode Viewer all the Renderer options

What happened instead?
a default value is used insted :D

Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-668-g3ebd54d (devel branch)

Printer model & used firmware incl. version
(if applicable - always include if unsure):
n/a

Browser and Version of Browser, Operating 
System running Browser (if applicable - always 
include if unsure):
Chrome OSX

Link to octoprint.log on gist.github.com or pastebin.com
(ALWAYS INCLUDE AND DO NOT TRUNCATE):
n/a

Link to contents of terminal tab or serial.log on 
gist.github.com or pastebin.com (if applicable - always 
include if unsure or reporting communication issues AND
DO NOT TRUNCATE):
n/a

Link to contents of Javascript console in the browser 
on gist.github.com or pastebin.com or alternatively a
screenshot (if applicable - always include if unsure
or reporting UI issues):
n/a

Screenshot(s) showing the problem (if applicable - always
include if unsure or reporting UI issues):
n/a

closed 2015-05-13 by GitIssueBot

MoonshineSG, 2015-05-13:
updated to reopen

foosel, 2015-05-13:
Not reopening as this is a duplicate of #443

MoonshineSG, 2015-05-13:
oh. ok. hope it gets implemented soon.  thanks.