OctoPrint/OctoPrint #892: Chosing AUTO serial port ignores specified baud rate

C-o-r-E, 2015-05-14:
I've observed that when connecting to a printer and selecting AUTO for the port, even if the baud rate is specified, it uses 115200.

In my case I have the baud set to 250000. 

I am using Octoprint from the devel branch which is at most 2 days old.

[Here](https://gist.github.com/C-o-r-E/52545f697907644ab4f3) is a snippet of the Octoprint terminal (not that its really needed).

C-o-r-E, 2015-05-14:
Hey @GitIssueBot 

I edited the issue. I would like to request a reopen.

ffr2822, 2015-05-28:
Some further information: I just set up OctoPrint using an octopi image - and found I could not connect to my TAZ 5 with "AUTO" - it never seemed to step down from 250Kbps. I had to specify 115Kbps. This matches what I would expect on a PI as the serial performance (even on a 2) is not exactly stellar.
Again, this was a TAZ 5, RAMBo 1.3L, Marlin 2015-Q1, Octoprint version 1.1.1-30

foosel, 2015-05-28:
@ffr2822 that's a different issue and from the sounds of it already fixed in devel

ffr2822, 2015-05-28:
No worries - just trying to add a data point.

Michael Baumann
baumann@uber-geek.org

"The most  exciting phrase to hear in science, the one that
 heralds new discoveries, is not 'Eureka!' (I found it!) but
'That's funny ...'" -- Isaac Asimov

foosel, 2015-05-29:
Should be fixed in `devel`

foosel, 2015-05-29:
@ffr2822 I think I came off a bit aggressive there, sorry for that. I just re-read your issue, and I'm not so sure anymore that it's what was fixed in `devel` or if it's something else entirely (or maybe even a problem with pyserial itself). Could you try how your TAZ behaves under `devel` and if the issue persists there [open a ticket](https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report)?

commit f2192ba2 (2015-05-29): Fix: Re-open detected serial port with specified baudrate

The serial port was previously kept open from the Stk500 initialization, which defaults to 115200 baud.

Closes #892
closed 2015-05-29 by foosel