OctoPrint/OctoPrint #895: Rotate USB cam

oskarsjogren, 2015-05-15:
Hi.

If I'm not mistaken, the MPJG plugin cannot rotate the webcam stream for USB cameras, or I'm not smart enough to find how to do it. Is this something that can be added to the web interface, in order to get nice time lapses even if the camera is mounted in the wrong way?

Thanks in advance
Best regards

markwal, 2015-05-23:
avconv (used in the octopi image) is capable of this via the transform filter.

Something like:
if settings().getBoolean(["webcam", "rotate90"]):
    filters.append('transpose=1')
on or around timelapse.py:284

That would do rotate the image in the timelapse.  On the control tab its a bit trickier because of css height, but I put an example rotate90 in the plugin https://github.com/markwal/OctoPrint-SnapStream, but it can only rotate the control webcam image.  Without replacing the whole timelapse module there isn't an easy way to add that filter to the ffmpeg command line (that I could see).

markwal, 2015-05-23:
And of course, rotate90 combined with fliph and flipv gives you 180 and 270 too.

foosel, 2015-05-29:
Implemented in `devel` by @markwal (#906)

closed 2015-05-29 by foosel