OctoPrint/OctoPrint #898: [REQUEST] Add raspberry pi core temperature to graph display... or at least somewhere in the GUI

bolsoncerrado, 2015-05-19:
ie log the activity of /opt/vc/bin/vcgencmd measure_temp
for raspi owners ;)

Thanks!

imrahil, 2015-06-18:
I can add this value to NavBar Temp plugin :)
is there any way to call system commands like this one for temp? subprocess.Popen or sarge.run?

foosel, 2015-06-18:
`sarge.run`

But for the sake of cross platform compatibility make it optional ;)

imrahil, 2015-06-18:
sure. I found nice platform detection script from Adafruit - https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/Platform.py

bolsoncerrado, 2015-06-18:
What plugin is that? ;)

imrahil, 2015-06-18:
http://plugins.octoprint.org/plugins/navbartemp/

bolsoncerrado, 2015-06-18:
Wohoo! I just realized on the new changes on the RCs and the plugin system! ;) good work!

foosel, 2015-06-30:
Provided through [the NavbarTemp plugin](https://github.com/imrahil/OctoPrint-NavbarTemp) by @imrahil  (also on the plugin repository)

closed 2015-06-30 by foosel