OctoPrint/OctoPrint #904: [Request] Rename locally stored files

quillford, 2015-05-22:
It would be useful if the file manager had the ability to rename gcode that was uploaded.

spapadim, 2015-07-31:
Could this be done in a plugin?  Without replicating the entire files tab, that is (i.e., have the plugin inject a template into each `div.entry` or, better, `div.entry > div.btn-group` of the file list).   

If so, I'd be interested to try out a couple of other ideas for file management...  A quick pointer to the right place in the code would help.  Thanks!

foosel, 2015-07-31:
It might be doable in a plugin, but right now I'm unsure. What would probably be better in any case (other than stuffing even more buttons into that poor narrow file list) would be a dedicated file manager. So the file list would stay as is, but you'd have a different tab page or settings dialog, or altogether different dialog that would allow you such things. I think it was @Salandora who once was working on something similar,  before there was a plugin system, but I never could get his adjustments to work :/

Salandora, 2015-08-02:
Yeah right I had this done before the plugin system, but until now I didn't found the time to convert it, it will be the "next" big project I'll do ;-)
I guess I can start this Week, base on how good and fast I get my Computer back up.

robseee, 2015-11-23:
+1 for rename support

Salandora, 2015-11-24:
So a little statement here: 
I'm working on the FileManager for a while now, as some of you might have noticed Octoprint now has a much better folder support (devel branch) which was/is essential for the file manager.
Actually I'm working on some bugs and fine tweaks before releasing anything.

bj0, 2016-07-15:
It would be nice to be able to rename a file that you want to keep or identify.  This is especially true if you use Slic3r's 'send to printer' option, which doesn't give you a filename option and tends to overwrite already existing files.

foosel, 2017-01-27:
File manager plugin is released, OctoPrint 1.3.0 is too, both together solve this. Closed.

closed 2017-01-27 by foosel