OctoPrint/OctoPrint #907: [Request] Disable/enable M105 temperature updates

MaxOLydian, 2015-05-25:
Is it possible to temporarily suspend the issuing of M105 temperature updates?

Under normal circumstances, when I have a connection open to my printer, OctoPrint sends M105 requests every few seconds to update the temperature graph.

If I send a long-running GCODE to the printer from the terminal frame, it seems that the M105 requests stack up and eventually overflow the command buffer.  When the currently executing command terminates, my printer is hung from the buffer overflow.  If I disconnect OctoPrint from the printer right after issuing the long-running command (to stop the M105s) and then reconnect after, everything is fine.

I realize this is really a bug with my printer firmware (Marlin Ramps 1.4), and I am submitting a bug report there, but might it not be nice to be able to turn off the M105 polling from OctoPrint?

Thanks for your consideration, and might I add a fantastic job on OctoPrint!

foosel, 2015-05-25:
Already solved in the 'devel' branch where no more than one M105 command will ever enter the send queue and known long running/blocking commands also are recognized properly.

closed 2015-05-25 by foosel

MaxOLydian, 2015-05-26:
Ack- sorry for the spurious request!  I hunted around a bit in the issues and didn't see anything related.  I'll grab the latest and give it a try.  Thanks!