OctoPrint/OctoPrint #909: Timelapses not rendering

stevelucky, 2015-05-26:
I'm having an issue where my timelapses aren't rendering after a print. 

1. What were you doing? Printing like normal

2. What did you expect to happen? A timelapse film to render at the end of my print.

3. What happened instead? An error appeared in OctoPrint that said:

Rendering of timelapse FILENAME.mpg failed with return code 1

/home/pi/.octoprint/timelapse/tmp/tmp_%05d.jpg: No such file or directory﻿

The log said:

2015-05-16 13:43:19,235 - octoprint.timelapse - WARNING - Could not render movie, got return code 1: /home/pi/.octoprint/timelapse/tmp/tmp_%05d.jpg: No such file or directory

4. Branch & Commit or Version of OctoPrint: Currently running 1.2.0-dev-854-gfa57f16 (devel branch) but I've had this issue for the last several versions. I've also completely wiped and recreated OctoPi and the issue persists.

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure): MakerFarm Prusa 10" i3v. Most current firmware, but I don’t have any numbers available at the moment.

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure): OX X Yosemite 10.10.4, Firefox 38.0.1

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

http://pastebin.com/raw.php?i=QPb9CfUq

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

serial.log is empty

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

http://pastebin.com/raw.php?i=c2nmfGCH

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues): I'm running a print right now and will attach a screenshot of the error as soon as it's done.

stevelucky, 2015-05-26:
I'm currently running a print and when I SSH into the ~/.octoprint/timelapse/tmp directory, it's still empty even though I should have several Z change snapshots in there.

foosel, 2015-05-26:
Have you tried with a time timelapse? Might be that OctoPrint does not correctly detect your z-changes and that this is the actual issue.

stevelucky, 2015-05-26:
This is a screenshot of the error.
[image]

I
l'll start another print right now that is timed instead of z change.

stevelucky, 2015-05-26:
I'd like to request a reopen. I'm not sure what info is missing as I read the FAQ and gave all the information that was asked for.

foosel, 2015-05-26:
I was just about to click reopen for these exact reasons. A crucial line from the original post you since added was missing which triggered it ;)

stevelucky, 2015-05-26:
Okay, that's what I was hoping. I noticed I missed that line. :-)

stevelucky, 2015-05-26:
I just started a timed timelapse (every 20 seconds) and snapshots are appearing in the ~/.octoprint/timelapse/tmp directory. The print won't be done for a while so I can't say whether or not it's going to render, but the fact that snapshots are appearing is good. So, it sounds like you were right, OctoPrint is not detecting z changes. Any idea why that could be or how to fix it?

foosel, 2015-05-26:
I just tried with my current devel build, and there z triggering still appears to work just fine. So maybe it's caused by files you are trying to print in some way. Could you upload a file with which the issue can be reproduced somewhere and provide the link? If you don't want to do this publicly, you can also mail me the link to osd@foosel.net

stevelucky, 2015-05-26:
Here's what I'm printing right now. 
https://www.dropbox.com/s/zpojagdxyembxdi/Mandible-SUPPORT_fixed-PLA.gcode?dl=0

It should be noted that this happens with every single I print. I'm using Slic3r version .9.9

stevelucky, 2015-05-26:
Yes, a timed timelapse worked great. It's just the z-change timelapse that isn't.

foosel, 2015-05-26:
I think I found and fixed it (see above commit). Please verify.

stevelucky, 2015-05-26:
Hmmm. I'm just having a little trouble finding those two files so that I can change them. What is their full path?

stevelucky, 2015-05-26:
Or do I just need to perform an update?

foosel, 2015-05-26:
Just perform an update.

stevelucky, 2015-05-26:
Hmmm. Now I can't connect to octopi.local at all. I can still SSH in so I know it's up, but the GUI won't appear. I just get:

503 Service Unavailable

No server is available to handle this request.

foosel, 2015-05-26:
Something went wrong while updating then, or you did forget to (re)start the server. Take a look into the log: `~/.octoprint/logs/octoprint.log`.

(I assume you followed [this](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image))

stevelucky, 2015-05-26:
Yes, I followed it exactly. Here's my octoprint.log

http://pastebin.com/raw.php?i=WWnwzRZ9

foosel, 2015-05-26:
This is still trying to start up your old version (note the version it prints during startup: `Starting OctoPrint 1.2.0-dev-854-gfa57f16-dirty (devel branch)` - the "-854" there should rather be a "-862", definitely not the same version you reported this ticket on). I also notice that your copy is dirty. Try the following (note that this will throw away all your changes, so if you still want them, save them somewhere first):

```
cd ~/OctoPrint
git reset --hard
git pull
~/oprint/bin/python setup.py clean
~/oprint/bin/python setup.py version
```

Verify that this prints out something along the lines of

     Version is currently: 1.2.0-dev-862-g3ac2d5d

If yes, continue

```
~/oprint/bin/pip uninstall --yes OctoPrint
~/oprint/bin/python setup.py install
sudo service octoprint restart
```

stevelucky, 2015-05-26:
This is what I'm getting:

>Version is currently: 1.2.0-dev-854-gfa57f16-dirty

stevelucky, 2015-05-26:
Also, when I do git pull, this is what I get:

>pi@octopi ~/OctoPrint $ git pull
Updating fa57f16..3ac2d5d
error: unable to unlink old 'docs/modules/index.rst' (Permission denied)
error: unable to create file docs/modules/server.rst (Permission denied)
error: unable to unlink old 'docs/plugins/hooks.rst' (Permission denied)
error: unable to unlink old 'src/octoprint/server/util/__init__.py' (Permission denied)
error: unable to unlink old 'src/octoprint/server/util/tornado.py' (Permission denied)

foosel, 2015-05-26:
Is this a current nightly of OctoPi?

stevelucky, 2015-05-26:
It's from an update sometime last week.

foosel, 2015-05-26:
Yeah, but was it originally a nightly or a regular release version of OctoPi? If the former, it might be that there's an issue with the build that @guysoft needs to know about. 

Anyway, your user can't write the update since it can't overwrite some of the changed files and so it ran into some very odd partial update situation. Try this:

    sudo chown -R pi.pi ~/OctoPrint

This should fix the permissions hopefully. Then try the update steps again.

stevelucky, 2015-05-26:
Yes, I downloaded the 2015-02-16-octopi-wheezy-0.12.0.img from the nightlies. That permission fixed seems to have fixed it. I did that then re-ran the update and now it's showing: Version: 1.2.0-dev-862-g3ac2d5d (devel branch) 

Also, it seems to have fixed the z change issue. screenshots are showing up in the ~/.octoprint/timelapse/tmp directory as I'm printing.

foosel, 2015-05-26:
Ok, great, then this is fixed.

closed 2015-05-26 by foosel