OctoPrint/OctoPrint #911: [Request] Printer response from /api/printer/command

ctag, 2015-05-26:
The API currently doesn't support querying the location of a printer's extruder/tool. 

My Marlin firmware implements the M114 command to return print head location. I hoped to use it with the [arbitrary command api](http://docs.octoprint.org/en/master/api/printer.html#send-an-arbitrary-command-to-the-printer) as a work around, but the POST request returns an empty body upon success.

Is it feasible to return the printer's response to a command via the /api/printer/command POST method?

Thanks,
ctag

foosel, 2015-05-27:
No, it's not feasible.

First of all, the communication with the printer happens asynchronously. The POST would thus have to block until the reply is received from the printer which could take a couple of seconds. Due to the synchronous nature of WSGI applications that would mean that the server would not handle requests until that has happened.

Secondly, most Marlin versions out there still have a bug that causes them to not respond to commands in the same order they were received in, so it's simply impossible for OctoPrint to match up any commands with their output. I pushed a patch to solve this around new years, so that should solve itself over time, but at the present time if capturing the output were to be implemented it would be completely unreliable, wrong most of the time and such useless. I'm not in the habit of implementing functionality that is bound to not work.

closed 2015-05-27 by foosel

silasb, 2016-09-10:
Is this still the case?  I just ran into this issue, but would love to workaround it somehow.

foosel, 2016-09-11:
Yes, this is still the case.