OctoPrint/OctoPrint #913: run specific commands on print start and finish

aerickson, 2015-05-28:
Could be used to change a light, play a sound, etc.

Perhaps different commands for print success or print failure (does Octoprint know this?).

closed 2015-05-28 by GitIssueBot

MolarAmbiguity, 2015-05-28:
@aerickson you might want to take a look at this https://github.com/foosel/OctoPrint/wiki/Cookbook:-Useful-Eventhooks. There also a plugin by @foosel for internet of things integration.

aerickson, 2015-05-28:
Eventhooks seem like the perfect solution. I was looking in the web UI and didn't see any way to do this. 

Thoughts about exposing the configuration of events in the webUI?

foosel, 2015-05-28:
It's on the TODO list as a plugin, but @Salandora already laid claim to that ;) so I'm keeping my feet still.

Depending on what you want to do, taking a look at [plugins](http://docs.octoprint.org/en/devel/plugins/gettingstarted.html) and especially the [EventHandlerPlugin](http://docs.octoprint.org/en/devel/plugins/mixins.html#eventhandlerplugin) might even be better than event hooks.