OctoPrint/OctoPrint #921: [Request] STL viewer tab

tharvey, 2015-06-01:
I love the Gcode Viewer tab and am thinking for those who are uploading stl and slicing with cura, why not a STL viewer tab to go along with it?

Poking around, I've found several javascript 3D model viewer and STL/OBJ parers that look like they work very well. Has this been discussed at all in the past?

I'm pretty new to 3D printing and Octoprint. I'm not sure how many users are uploading and slicing STL's on Octoprint directly. I started off this way, but after realizing that I like to edit the extrusion rates per-model and not wanting to constantly edit my profile in Octoprint I've started slicing externally and uploading the gcode.

foosel, 2015-06-17:
> I've found several javascript 3D model viewer and STL/OBJ parers that look like they work very well. Has this been discussed at all in the past?

Yes. Too much for 1.2.0, definitely a TODO for a later release though, or even better, as a plugin.

jneilliii, 2015-06-17:
I like the idea of making this a plugin.  I think I might try my hand at programming python and see how it works out.  I have extensive experience with web/vb programming, can't be that difficult right?

foosel, 2015-06-17:
Well, at least I tried to make it as straight forward as possible :) There's a getting started guide in the docs, I recommend starting with that :)

jneilliii, 2015-06-19:
The HelloWorld example was great for getting started.  Here's my starting skeleton for developing this plugin.

https://github.com/jneilliii/Octoprint-STLViewer

jneilliii, 2015-06-19:
Anybody have ideas on how to read in a list of uploaded files, filtered by stl only, and generate a select list.  Looking through the documentation it seems there are [inherited properties](http://docs.octoprint.org/en/master/plugins/concepts.html#injected-properties) that would allow using the octoprint.filemanager.FileManager but there's no documentation on that object in the docs.  Is this the same as octoprint.filemanager.storage and I could data-bind to the list_files method?

markwal, 2015-06-19:
In your ViewModel, you can retrieve the list of all local files via ajax and then filter that list down to only stl based on type.  If you store the result of that in a member variable in your ViewModel, you can easily bind that into your select via knockout.

markwal, 2015-06-19:
The REST API for files is here: http://docs.octoprint.org/en/master/api/fileops.html.

foosel, 2015-06-19:
I'd actually suggest having the `FilesViewModel` injected into your custom viewmodel, that will also take care of updating itself on upload events and such things.

Source is [here](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/static/js/app/viewmodels/files.js), sorry, no docs yet - and please ignore that it's called `GcodeFilesViewModel` still, haven't renamed it yet. That contains all files retrieved from the API in `self.listHelper` for which the source can be found [here](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/static/js/app/helpers.js#L1). Basic gist: `self.listHelper.items()` will give you an observable array you can bind to. You want those entries whose `type` attribute reads `model` ([binding example](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/templates/sidebar/files.jinja2#L21)).

Also I have the sneaking suspicion right now I forgot to update parts of the file api docs, the type attribute is not listed in there. Gna...

markwal, 2015-06-19:
Oh yeah.  That's way better because the files view model already has the list you'll save some memory and round trips in the app.

jneilliii, 2015-06-19:
Thanks a lot.  This part will probably take me a while as I figure out knockout.  It seems pretty straight forward though.

jneilliii, 2015-06-20:
So I have a working viewer now on my [master branch](https://github.com/jneilliii/Octoprint-STLViewer), the only issues I'm stuck on is how to filter/sort the list and it doesn't seem to dynamically update when I upload new files.

foosel, 2015-06-30:
Provided through [the STLViewer plugin](http://plugins.octoprint.org/plugins/stlviewer/) by @jneilliii

closed 2015-06-30 by foosel