OctoPrint/OctoPrint #927: Plugin Manager doesn't appear in the settings

tonyphoang, 2015-06-08:
Hello, I'm running a fresh install of Octopi and noticed there wasn't a plugin manager link in my settings. It says in the user guide that the plugin manager comes pre-bundled with Octopi. Can someone comment on this? Thanks.

closed 2015-06-08 by GitIssueBot

markwal, 2015-06-08:
It's brand new, so you probably switched branches to devel before it was there. Update to the latest: https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image

tonyphoang, 2015-06-08:
I attempted to follow the instructions listed but got this error

pi@octopi ~/OctoPrint $ source ~/oprint/bin/activate
(oprint)pi@octopi ~/OctoPrint $ git pull
error: Unable to find dfecbac51923a01edd05402f0dd0a86c56824e51 under http://gnet.homelinux.com/git/OctoPrint.git
Cannot obtain needed tree dfecbac51923a01edd05402f0dd0a86c56824e51
while processing commit 57de36a9d3fdded45329d02a6567eff7976ebd47.
error: Fetch failed.
(oprint)pi@octopi ~/OctoPrint $

foosel, 2015-06-08:
@guysoft this looks like the recent build of OctoPi is broken regarding the preset origin remote, I suggest removing it until your build server is back in action

@tonyphoang 

    cd ~/OctoPrint
    git remote set-url origin https://github.com/foosel/OctoPrint
    git pull
    ~/oprint/bin/python setup.py install

tonyphoang, 2015-06-08:
That seemed to work. Could you update the wiki for it?

foosel, 2015-06-08:
No, because this should normally not be necessary and that it is is a bug in the OctoPi build you are running, see above