OctoPrint/OctoPrint #931: Can't install plugins

Findus23, 2015-06-10:
1. What were you doing?

I wanted to install https://github.com/OctoPrint/OctoPrint-Pushbullet/commit/50e8cea679e2f638180593b6f484b66e2ed50cc4 via GUI and pip.
2. What did you expect to happen?

Plugin installs successfully
3. What happened instead?

GUI: the following not helpful error occurs:
>Etwas ist schief gegangen
>Installation des Plugins von URL "unknown" fehlgeschlagen: Could not parse output from pip

pip:
```bash
pi@octopi ~/OctoPrint $ ~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
```
```
Downloading/unpacking https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
  Downloading master.zip
  Running setup.py egg_info for package from https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
    Traceback (most recent call last):
      File "<string>", line 14, in <module>
      File "/tmp/pip-Pqvac6-build/setup.py", line 16, in <module>
        "pushbullet.py"
      File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_931_g06fb249-py2.7.egg/octoprint_setuptools/__init__.py", line 494, in create_plugin_setup_parameters
        packages = find_packages(where=source_folder, exclude=ignored_packages, include=[package, "{package}.*".format(**locals())] + additional_packages)
    TypeError: find_packages() got an unexpected keyword argument 'include'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 14, in <module>

  File "/tmp/pip-Pqvac6-build/setup.py", line 16, in <module>

    "pushbullet.py"

  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_931_g06fb249-py2.7.egg/octoprint_setuptools/__init__.py", line 494, in create_plugin_setup_parameters

    packages = find_packages(where=source_folder, exclude=ignored_packages, include=[package, "{package}.*".format(**locals())] + additional_packages)

TypeError: find_packages() got an unexpected keyword argument 'include'

----------------------------------------
Command python setup.py egg_info failed with error code 1 in /tmp/pip-Pqvac6-build
Storing complete log in /home/pi/.pip/pip.log
```
4. Branch & Commit or Version of OctoPrint:

https://github.com/foosel/OctoPrint/commit/06fb249f2e1e78651d84da8080946e64351a134b
5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
K8200
6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Chromium 43.0.2357.65 (most likely browser independent)
7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
https://gist.github.com/Findus23/1c887c370ecf1c6b5a28
I have read the FAQ.

Sorry for opening the issue in the wrong place. (https://github.com/OctoPrint/OctoPrint-Pushbullet/issues/1)

foosel, 2015-06-10:
I was just about to comment at the other ticket that this is an OctoPrint issue ;)

You should be able to fix this via `~/oprint/bin/pip install setuptools==7.0`, I'll push a fix shortly (would have been the first RC ever without any issues ;))

closed 2015-06-10 by foosel

Findus23, 2015-06-10:
Thanks for that quick fix.
Now I can install https://github.com/Salandora/OctoPrint-SystemCommandEditor/commit/a910402696247029dcef5984f19b774101af0985  (but I had to force it with --upgrade).
But it seems as there is a second issue with OctoPrint-Pushbullet::
```
pi@octopi ~ $ ~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
Downloading/unpacking https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
  Downloading master.zip
  Running setup.py egg_info for package from https://github.com/OctoPrint/OctoPrint-Pushbullet/archive/master.zip
    
Requirement already satisfied (use --upgrade to upgrade): OctoPrint in ./oprint/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_931_g06fb249-py2.7.egg (from OctoPrint-Pushbullet==0.1.0)
Requirement already satisfied (use --upgrade to upgrade): requests in ./oprint/lib/python2.7/site-packages (from OctoPrint-Pushbullet==0.1.0)
Downloading/unpacking pushbullet.py (from OctoPrint-Pushbullet==0.1.0)
  Running setup.py egg_info for package pushbullet.py
    Traceback (most recent call last):
      File "<string>", line 14, in <module>
      File "/home/pi/oprint/build/pushbullet.py/setup.py", line 15, in <module>
        install_reqs = parse_requirements("requirements.txt",  session=uuid.uuid1())
    TypeError: parse_requirements() got an unexpected keyword argument 'session'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 14, in <module>

  File "/home/pi/oprint/build/pushbullet.py/setup.py", line 15, in <module>

    install_reqs = parse_requirements("requirements.txt",  session=uuid.uuid1())

TypeError: parse_requirements() got an unexpected keyword argument 'session'

----------------------------------------
Command python setup.py egg_info failed with error code 1 in /home/pi/oprint/build/pushbullet.py
Storing complete log in /home/pi/.pip/pip.log
```

foosel, 2015-06-10:
You had to force although you did provide the `==7.0` part as well? Shouldn't be the case...

The error above now in turn is caused by a 3rd party dependency of the pushbullet plugin, so let's discuss further in OctoPrint/OctoPrint-Pushbullet/#1 :)

adammhaile, 2016-09-27:
This is back as of 1.2.16 for me. Running on Pi2 from the OctoPi distro. Cannot install any plugins.

[plugin_pluginmanager_console.txt](https://github.com/foosel/OctoPrint/files/496544/plugin_pluginmanager_console.txt)

foosel, 2016-09-27:
Your log begs to differ and contains several "installed successfully" messages. And it also does not contain the error that was present in this issue so it's not the same problem in any case. Red herring :)

Please open a new ticket and fill out the full template. Also include screenshots showing the problem (apparently you are coming to the conclusion the install failed even though the provided log says otherwise, so my guess is there's some message on your screen that indicates file) and octoprint.log. Also state from what version you upgraded to 1.2.16, since 1.2.15 to 1.2.16 contains absolutely no changes in the plugin manager. Give as full of a picture as possible.

commit 6a7a8580 (2015-06-10): Fix: octoprint_setuptools depends on setuptools==7.0

Closes #931