OctoPrint/OctoPrint #932: Cura Slicing: Return Code 2

nacho693, 2015-06-12:
1. What were you doing?

Slicing with CuraEngine--new install

2. What did you expect to happen?

That it would slice

3. What happened instead?

Return Code 2

4. Branch & Commit or Version of OctoPrint:

Version: 1.2.0-rc1-2-g8353c9b (master branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

M2 MakerGear

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Multiple

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

http://pastebin.com/VcuZ7vyE

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

N/A BUT plugin_cura_engine.log : http://pastebin.com/KTq9HMPw
10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

foosel, 2015-06-12:
Wrong Cura (at least from how I read your cura_engine.log). As stated in the change log the plugin uses CuraEngine directly instead of the old wrapped approach. Point the plugin to the CuraEngine binary instead please.

closed 2015-06-12 by foosel

nacho693, 2015-06-12:
You are correct sir, thank you.