OctoPrint/OctoPrint #933: Default slicing profile

mic159, 2015-06-15:
The plugin API provides a default slicing profile, but when you first run octoprint, there is no slicing profiles listed.

mic159, 2015-06-17:
Updated ticket to comply. Can this be reopened?

foosel, 2015-06-17:
> if you have not saved any profiles yet, there is no way to slice (no option to use the default profile) [...] I was wondering, is this intentional or a bug/unfinished feature?

Intentional behaviour. Allowing slicing just based on the default profile would in 90% of all cases produce hilariously wrong gcode for the connected printer. To save the users from having to discover that on their own (and mass occurences of related tickets here that I could only close with "well, then configure it properly..."), it won't let you slice unless there's a slicing profile. The fact that there even is a default slicing profile available is only to provide defaults for the less sensitive parameters like defaults for support, platform adhesion, extruder account (1 is probably never wrong) etc. Stuff like filament diameter, bed size etc should always be provided by the user who knows his machine best.

PS:
> Stuff to keep non-humans happy:

Nope, also the humans. You wouldn't want to have to run after every single person opening a ticket to provide some basic information like use version and logs either, right? ;) After doing that to me for >18 months I decided I needed a bot...

mic159, 2015-06-17:
> Allowing slicing just based on the default profile would in 90% of all cases produce hilariously wrong gcode for the connected printer.

Yep, makes sense.

Building templates, JS, CSS and blueprints just to allow setting up slicing profiles for my plugin is a non-zero amount of effort. Just going to inject some default profiles on startup if it doesn't exist for now :D
There aren't many settings for my laser engraver, just speed and some raster options (B/W vs greyscale).

> Nope, also the humans.

Totally understand

closed 2015-06-17 by mic159

foosel, 2015-06-17:
> Just going to inject some default profiles on startup if it doesn't exist for now

Totally valid approach :) Just put them as additional package data into the plugin and copy them over on startup if they are not there yet :)

mic159, 2015-06-17:
Have you thought about allowing some sort of UI-building like the custom controls for creating plugin settings?

foosel, 2015-06-17:
Somewhat.. there's still a biiiiig PR in the pipeline for a complete refactoring of how the settings work, and allowing some generated forms for plugin settings through that too would be a logical consequence.