OctoPrint/OctoPrint #934: Problems running Webcam with ID 1e4e:0110

Barracdil, 2015-06-17:
Hi :)

sorry If I am wrong here - But I dont know where else I could post. I want to set up OctoPrint in combination with a Webcam (lsusb ID 1e4e:0110). Regarding the Webcam compability list it should work with the HAMA Webcam Settings

Quote:
Hama PC-Webcam "AC-150" on Raspberry Pi, tested by StefanBurger (use ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -y -r 640x480 -f 10 to start)

If I try to run: ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -y -r 640x480 -f 10
over the terminal in the mjpg folder I get ERRORs:

mjpg_streamer: unrecognized option: "-w"

When I using the string without -w I get error unkown -y and then unknown -r

I tried using the last stable OctoPI Image and the latest nightly (15.06.15) without success. What I am doing wrong? To be honest - I am not that Linux Crack - My skills are very very limited ;)

closed 2015-06-17 by GitIssueBot

foosel, 2015-06-17:
You are wrong here, this would be better located over in the [issue tracker of OctoPi](https://github.com/guysoft/OctoPi/issues), but still, try this:

  1. edit `/boot/octopi.txt` (e.g. `nano /boot/octopi.txt`, also see [this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-change-the-webcam-resolution-on-octopi))
  2. remove the `#` in front of the `camera_usb_options="-r 640x480 -f 10"` line
  3. change that line to `camera_usb_options="-y -r 640x480 -f 10"`

The whole file should look like this:

```
### Configure which camera to use
#
# Available options are:
# - auto: tries first usb webcam, if that's not available tries raspi cam
# - usb: only tries usb webcam
# - raspi: only tries raspi cam
#
# Defaults to auto
#
#camera="auto"

### Additional options to supply to MJPG Streamer for the USB camera
#
# See https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration
# for available options
#
# Defaults to a resolution of 640x480 px and a framerate of 10 fps
#
camera_usb_options="-y -r 640x480 -f 10"

### additional options to supply to MJPG Streamer for the RasPi Cam
#
# See https://github.com/foosel/OctoPrint/wiki/MJPG-Streamer-configuration
# for available options
#
# Defaults to 10fps
#
#camera_raspi_options="-fps 10"
```

OctoPi will take care of starting up mjpg_streamer with these parameters then on the next reboot. 

Also a word of warning, your camera does not support mjpg out of the box apparently (or you wouldn't need the -y parameter). This means that your Pi will have to do the transcoding from the yuv data stream into mjpg, which puts additional strain on the CPU and can negatively impact print quality. I strongly suggest to get a camera that natively support mjpg, e.g. the Logitech C270.

Barracdil, 2015-06-17:
Best thanks foosel! So easy - when you know it :) But you are right. The CPU load is very high. Will looking for a alternative then :) Thanks!