OctoPrint/OctoPrint #935: [Brainstorming] Allow slicers to support other file formats

mic159, 2015-06-17:
I think it would be good to allow slicers to extend the list of allowed file extensions that can be uploaded and sliced.

For now I have local mods to add `.png` to the file manager as a "model" type [here](https://github.com/foosel/OctoPrint/blob/2e72b91eaf3ca99446a0c15fe70f701ccb6cd0fd/src/octoprint/filemanager/__init__.py#L28)
and disabled the file extension checking on the slice endpoint [here](https://github.com/foosel/OctoPrint/blob/2e72b91eaf3ca99446a0c15fe70f701ccb6cd0fd/src/octoprint/server/api/files.py#L325)

1. What were you doing?
Trying to slice a PNG image into GCODE for engraving.

2. What did you expect to happen?
Be able to upload the PNG image to benbox and slice it.

3. What happened instead?
Wont allow upload, because file extension is not ".stl"

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-rc1-3-g46711ce-dirty (master branch)

5. Printer model & used firmware incl. version
Benbox Laser Engraver
`benbox laser 2.7(c) mul 2014.12.25`

6. Browser and Version of Browser, Operating System
Chrome 42, Ubuntu

7. Link to octoprint.log on gist.github.com or pastebin.com
   ...

8. Link to contents of terminal tab or serial.log on 
   ...

9. Link to contents of Javascript console in the browser 
   ...

10. Screenshot(s) showing the problem
  ...

I have read the FAQ.

foosel, 2015-06-17:
Apart from [this](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-extension-tree)?

mic159, 2015-06-17:
Ooo, I missed that! Will give it a shot tomorrow.
Thanks!

mic159, 2015-06-20:
I took a look at the extension hook and it works well to allow it to be uploaded, but you still cant slice the new format because of this: https://github.com/foosel/OctoPrint/blob/2e72b91eaf3ca99446a0c15fe70f701ccb6cd0fd/src/octoprint/server/api/files.py#L325

The filetype I'm slicing is `png`.
```python
def png_file_support(*args, **kwargs):
    return {
        'model': {
            'png': ['png']
        }
    }
```
The only workaround I can think of is registering it as an alternative extension for 'stl' which is a very dirty hack.

I think it would be better if the file types supported was part of the slicer properties so that the interface can show slicers that support that particular file type.

foosel, 2015-06-20:
Ok, noted, I'll try to get that into the next release after 1.2.0

mic159, 2016-07-02:
still not possible :(

foosel, 2016-07-02:
@mic159 You are right, I totally forgot about this. To make up for it I used some of my weekend just now and whipped something up. Could you take a look at the `improve/slicingFlexibility` branch and see if it works for you?

mic159, 2016-07-03:
@foosel Thanks!
Had a quick go at it, but unfortunately it gives me a `HTTP 415 UNSUPPORTED MEDIA TYPE` when I attempt to start the slicing.

I can upload the png, and it has the slice button, lets me set the options, but when I press "Slice", it says 415.

My plugin can be found here: https://github.com/mic159/benbox-slicer/blob/feature/octoprint_filetype/octoprint_benbox_slicer/plugin.py

foosel, 2016-07-16:
Merged (a while ago *cough*), closing

closed 2016-07-16 by foosel