OctoPrint/OctoPrint #936: [Request] Confirm restart after installing plugin

Jerware, 2015-06-17:
I installed a plugin while printing (whoops). It would be nice if the user were given the option to restart later instead.

foosel, 2015-06-17:
Installing a plugin should not automatically restart. Do you mean restart or reload?

Jerware, 2015-06-17:
Really? I installed the Print History plugin (0.4). After installation it informed me that I needed to restart before it would take effect. I clicked OK, and then a few moments later my print stopped and Octoprint restarted. I'm running OctoPi Version: 1.2.0-dev-943-g9812232 (devel branch).

I'll try to reproduce the issue as soon as my current print finishes.

As an aside, I had a print inexplicably stop last night, so I changed my overclock settings from "Pi 2" (which I'm running) to "None". I made this change before this restart issue but thought I'd mention it since it's the only other weirdness I've experienced.

foosel, 2015-06-17:
I still have a big TODO to allow restarting OctoPrint automatically from the Plugin Manager when new plugins that need it are installed or uninstalled, and I definitely haven't tackled that yet... So unless I'm completely misunderstanding something here right now, it shouldn't have restarted, and if it did, that's a bug. If you happen to have a log file, that would help. I guess there isn't some additional plugin at works here that might be listening to plugin lifecycle events and cause a restart?

Also the shouldn't be anything to click "ok" on, the Plugin Manager only generates a "read-only" notification O_O

[image]

The Software Update plugin can restart automatically if configured with a restart command, but that doesn't allow you to trigger updates during a print actually.

Other than that only the automatic reload comes to mind that happens when it detects a change in the server version, but that is front-end only, so the server would keep running. It would be the same as hitting reload in your browser.

Jerware, 2015-06-17:
Of course, you're right about no OK button. Since I couldn't reproduce right away, I was basing my report on memory. Shameful bug reporting, I know. :)

Here's today's log, if it's helpful.

http://ledseq.com/downloads/octoprint.log

foosel, 2015-06-17:
Hm... not seeing anything suspicious in there. Tell you what. I'll mark this as a bug ticket, but will set it to unreproducible for now (since I certainly can't reproduce it). You'll try to reproduce it (or maybe someone else will come around with the same issue). If it doesn't happen again in let's say the next four weeks we'll consider it a case of some cosmic radiation and close it carefully, and if it happens after that, one ping and it's open again. Deal? :)

foosel, 2015-06-20:
Still haven't reproduced the original problem, but the plugin manager now refuses any actions while a print job is active and notifies the user in the UI accordingly:

[image]

[image]

See commit 0ce1575e12b2e66fe4db0e1cff8f068938ed1a14

Jerware, 2015-06-22:
I can't reproduce the issue either, so I apologize for the false alarm. I think it might have been a power issue because since I switched my power source I've had no more rebooting issues. I think it was just bad timing that caused me to think it was related to the plug-in install.

foosel, 2015-06-30:
Ok, closing this then.

closed 2015-06-30 by foosel