OctoPrint/OctoPrint #937: [Request] RTSP Support for IP cameras

wes1007, 2015-06-17:
Cannot set the RTSP link of an Axis IP camera under camera settings in the WEB GUI.

link would look like this: rtsp://192.168.1.103/axis-media/media.amp

I noticed someone else also wanting support [Here](https://groups.google.com/forum/#!topic/octoprint/NIaUCUckGaA)

Renha, 2015-07-08:
As a workaround you could hack your camera to install mjpg-streamer there. I am in process of doing that for Xiaomi Ants.

tdaede, 2015-08-09:
Browsers can't play RTSP streams, so I'm not sure how valuable this would be. Seems like this would be better solved via modifying the camera, or setting up some sort of mjpg-streamer like proxy.

ozett, 2015-09-25:
rtsp no only for webcam image in octoprint...
i would like to see getting timelapse videos from other cams than usb/raspi-cam.
i could be done with ffmepg like this:

ffmpeg -i $RTSPURL -y -f image2 -ss 8 -sameq -t 0.001 $CAMIMGPATH/$MONT/$DATE/$DATETIME.jpeg

would like to see the octoprint webif to support this option...

mjepg with ffmpeg-server is a lot of fiddling...

ozett, 2015-09-25:
this seems also a requstable feature:

http://www.makermusings.com/2015/09/01/making-time-lapse-video-of-3d-prints/
"put nozzle out of the way to take the screenshot"

maybe it could also be done with a "@execute-" postprocessor./script-trigger..? would be a fine feature in octoprint...

nonnos, 2015-10-05:
Renha whats your status with mjpeg-streamer for Xiaomi Ants?

Renha, 2015-10-11:
@nonnos sorry no big progress, Ants have too small amount of Ram to compile anything there and no swap support so i need virtual env, in which I have no much experience.

seewaldjan, 2017-01-08:
maybe someone got a workaround to maybe convert the stream?

MathijsG, 2017-09-13:
I use this shell script for streaming to YouTube. Just put in your livestream-code instead of PUTYOUTUBE-IDHERE :

`ffmpeg -re -f mjpeg -framerate 5 -i "http://localhost:8080/?action=stream" -ar 44100 -ac 2 -acodec pcm_s16le -f s16le -ac 2 -i /dev/zero -acodec aac -ab 128k -strict experimental -s 640x480 -vcodec h264 -pix_fmt yuv420p -g 10 -vb 700k -preset ultrafast -crf 31 -framerate 5 -f flv "rtmp://a.rtmp.youtube.com/live2/PUTYOUTUBE-IDHERE"

`

jneilliii, 2018-01-20:
My plugin allows for viewing YouTube live stream in a seperate tab of OctoPrint.  I may look into incorporating this command-line to enable starting the stream.

https://plugins.octoprint.org/plugins/YouTubeLive/

2andr, 2018-12-24:
Any news about RTSP stream in Octoprint ?

pyr0ball, 2020-01-07:
Been another year, just checking in if there's been any new support options?

Wyze cameras have released an official firmware that allows RTSP streams now, would be nice to be able to utilize those since they're dirt cheap and easy as pie to set up.

As it is now, I'll probably have to set up some kind of ffmpeg/mjpegstreamer instance to make this work

jneilliii, 2020-01-07:
You may want to check out this thread in the community forum. 

https://community.octoprint.org/t/wyze-cam-v2-support/6651/24?u=jneilliii

pyr0ball, 2020-01-08:
I'd already implemented an ffmpeg + ffserver solution, but this looks like it might be less overhead.

Anyhoo, it's all well and good for someone like me, who's experienced in linux backend, to do something like this. I just wish there was an easy front-end solution I could point people to.

Maybe I'll use this as an opportunity to learn how to make an octoprint plugin

pyr0ball, 2020-01-17:
@jneilliii Any chance you could help me debug using vlc for this? ffmpeg isn't playing very nice, and at the moment I'm not able to get the vlc command line to work either. Don't think this is an appropriate thread to continue conversation on, however

jneilliii, 2020-01-17:
You should go over to the community forum where it's mentioned for the VLC command line. I've never personally done it, but one if the posters over there did. 

https://community.octoprint.org/t/wyze-cam-v2-support/6651/24?u=jneilliii

hsunner, 2021-01-15:
Most (all?) axis cameras have MJPEG HTTP endpoints. This works for octoprint:
- http://[IP]/axis-cgi/mjpg/video.cgi
- http://[IP]/axis-cgi/jpg/image.jpg

Check out the full VAPIX documentation: http://www.axis.com/vapix-library or google "axis cgi endpoints" to find third-party summaries. Both scripts take a plethora of options (?color=0) and viewports etc can be configured (?camera=2)

Note that you need to "Allow anonymous viewers" ("System" tab | "Plain config" group | "Allow anonymous viewers"). In principle you should be able to provide a username/password for the http auth in the URL (http://user:pass@[IP]/axis-cgi/jpg/image.jpg), but octoprint does not seem to recognize this form of URI. (However, if you are logged into the web interface of the camera, you will have access to watch the mjpeg feed in the octoprint window if it's on the same browser.)

ozett, 2021-04-01:
still a workaround up to today to use RTSP/h264/h265 als live image: 
use motioneye/motioneye-os to convert rtsp to mjepg and than you can use also this plugin https://github.com/jneilliii/OctoPrint-MultiCamView or the octoprint mjpeg_stream with its daemon

jneilliii, 2022-01-23:
Steps to utilize restreamer for this...

https://gist.github.com/jneilliii/fb40f42d16887649b58699f002fe48ec