OctoPrint/OctoPrint #939: [Request] Update knockout.js to version 3.3.0

punkkeks, 2015-06-18:
For my current application it would be nice to use knockout in version 3.3 (because of [this bug][0]). I downloaded it and renamed it to knockout.js, restarted octoprint and everything runs like before. No errors in console, no visible crap.
I admit, that this needs further investigation, but at first glance, it works fine.

[0]: http://knockoutjs.com/upgrade-notes/v3.0.0.html#bindings-are-now-refreshed-independently

foosel, 2015-06-18:
I'm a bit worried about http://knockoutjs.com/upgrade-notes/v3.0.0.html#checked-binding-uses-strict-equality and would like to evaluate this a bit myself, so this is nothing I'll put into 1.2.0. However I'll put it on the TODO list for 1.2.1

foosel, 2016-02-02:
Updated in `maintenance` and `devel` branches, will become part of 1.2.9

commit caaf1d22 (2016-01-27): Updated Knockout dependency to version 3.4.0

Closes #939
closed 2016-02-02 by foosel