OctoPrint/OctoPrint #940: Cannot

Chiliyago, 2015-06-18:
1. What were you doing?
Opening Octoprint for the first time.

2. What did you expect to happen?
Connect to Printer automaticaly

3. What happened instead?
I received this message on the home page:
Machine State: Error: Failed to open serial port, permissions correct?

This is found in the Terminal 
Changing monitoring state from 'Offline' to 'Opening serial port'
Connecting to: None
Unexpected error while connecting to serial port: None OSError: '[Errno 2] No such file or directory: 'None'' @ comm.py:_openSerial:943
Changing monitoring state from 'Opening serial port' to 'Error: Failed to open serial port, permissions correct?'
Changing monitoring state from 'Offline' to 'Opening serial port'
Connecting to: None
Unexpected error while connecting to serial port: None OSError: '[Errno 2] No such file or directory: 'None'' @ comm.py:_openSerial:943
Changing monitoring state from 'Opening serial port' to 'Error: Failed to open serial port, permissions correct?'

4. Branch & Commit or Version of OctoPrint:
Version: 1.1.1-30-g4fede5a (master branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
Prusa i3.  I am not sure I just of firmware at this time

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

OctoPrint is running on latest verison of OctoPi dated 13-Fed-2015
I run all major & current browsers.  All report the same error.

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
https://gist.github.com/Chiliyago/5a3601c098756329ed09

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

The serial.log is empty

This is in the terminal windows.
Changing monitoring state from 'Offline' to 'Opening serial port'
Connecting to: None
Unexpected error while connecting to serial port: None OSError: '[Errno 2] No such file or directory: 'None'' @ comm.py:_openSerial:943
Changing monitoring state from 'Opening serial port' to 'Error: Failed to open serial port, permissions correct?'

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

I am not sure what this means.

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

Screenshot Here: http://1drv.ms/1N60QlY

I have read the FAQ.

foosel, 2015-06-18:
Your printer couldn't be autodetected.

Please try with current release candidate (current `master` branch). 1.1.1 is old (1.1.2 was current before 1.2.0-RC-1 and -2). If that doesn't help, try selecting your printer port manually from the Port drop down list. If your printer doesn't show up there, make sure the OS of the computer/pi running OctoPrint does detect it properly.

chris666uk1, 2015-06-20:
i have same problem was ok updated to Version: 1.2.0-dev-965-g65ae48d (devel branch) and i get 

Machine State: Error: Failed to open serial port, permissions correct?

foosel, 2015-06-20:
No, this is a different problem. You are running into an error because your version of pyserial doesn't support 250000 baud. This is nothing that can be fixed in OctoPrint.

If you are running OctoPi, please remove the pre-installed version of pyserial via sudo apt-get remove python-serial. That has been known to sometimes win over the viewer version installed in the virtual env on OctoPi.

chris666uk1, 2015-06-20:
removed now i get this 

i am running on a bananapi  and fresh build

Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/bin/octoprint", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: pyserial

chris666uk1, 2015-06-20:
thanks for help managed to fix it i did the following

apt-get remove python-serial
reboot

cd ~/OctoPrint/
git pull
git checkout devel
./venv/bin/python setup.py install

~/OctoPrint/venv/bin/octoprint

all working connected with no problems thanks for your help

foosel, 2015-06-23:
@Chiliyago Have you tried what I told you to try? Is it working now?

chris666uk1, 2015-06-23:
my above post fixed it for me

Chiliyago, 2015-06-24:
The owner of OctoPi provided a link to a new image. I installed that and it is now working.  Thank you @foosel

closed 2015-06-24 by foosel

kentmartin99, 2016-02-14:
Hi,
I am having similar issues with octoprint (v1.2.2). I am running on a raspberry pi with a Printrboard as the 3D printer controller. 
I get error "Error: Failed to autodetect serial port, please set it manually." 
I tried auto and manually setting baud rates, but nothing worked.

The octoprint terminal shows 
"Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Error while connecting to /dev/ttyACM0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually."

When plugging printer into usb port i get this from dmesg:
"[  135.416076] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[  941.313221] usb 1-1.3: USB disconnect, device number 6
[  946.153793] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
[  946.267227] usb 1-1.3: New USB device found, idVendor=16c0, idProduct=0483
[  946.267268] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  946.267288] usb 1-1.3: Product: USB Serial
[  946.267309] usb 1-1.3: Manufacturer: Teensyduino
[  946.267326] usb 1-1.3: SerialNumber: 12345
[  946.275985] cdc_acm 1-1.3:1.0: This device cannot do calls on its own. It is not a modem.
[  946.276129] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device"

I didn't upgrade to v1.2.9 because it killled it when I tried last time.

What should I do next?
Thanks.

foosel, 2016-02-14:
Auto detection doesn't work if the controller isn't arduino based. What happens if you just select ttyACM0 in the port drop down instead of leaving that on AUTO and then connect? Note that in order for that port to be selectable here, the printer needs to be plugged in *before* you load OctoPrint in your browser, if you plug it in after that you need to reload for it to show up.

kentmartin99, 2016-02-15:
All good now, it's working. Thanks.

foosel, 2016-02-15:
Alright. You should still update. If it "stopped working" that should be looked into instead of not updating, or you won't get any bug fixes and security updates. There is a mailing list and a G+ community where you can get support with problems like these.

Jpson67, 2016-04-23:
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.
Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.