OctoPrint/OctoPrint #941: The command "Turn on the printer" could not be executed.

MauroHKT, 2015-06-18:
Hi guys I don't know how to solve this problem, I've google everywhere but unable to find solution.
I want to use a Grove relay to Turn on-off my printer, I've followed guide:
https://github.com/foosel/OctoPrint/wiki/Controlling-a-relay-board-from-your-RPi
modified 2 files , restarted....
now I have the 2 buttons under "system" menu but if I try I get error:
2015-06-18 21:50:38,962 - octoprint.server.util - INFO - New connection from client: 192.168.2.1
2015-06-18 21:50:46,204 - octoprint.server.api - INFO - Performing command: gpio -g write 18 1
2015-06-18 21:50:46,224 - octoprint.server.api - WARNING - Command failed with return code 127: 
2015-06-18 21:56:00,750 - octoprint.server.api - INFO - Performing command: gpio -g write 18 1
2015-06-18 21:56:00,770 - octoprint.server.api - WARNING - Command failed with return code 127: 

[image]

closed 2015-06-18 by GitIssueBot

foosel, 2015-06-18:
Please address this on the Mailinglist or the G+ Community (both are linked from the Contribution Guidelines). This is a support question, not a bug in OctoPrint.

MauroHKT, 2015-06-18:
Ok, Many Thanks