OctoPrint/OctoPrint #946: Exception on /api/settings

mattsch, 2015-06-20:
Getting the following traceback with master/rc3 after updating today:

```python
2015-06-20 11:46:06,220 - octoprint - ERROR - Exception on /api/settings [GET]
Traceback (most recent call last):
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/OctoPrint-1.2.0_rc3-py2.7.egg/octoprint/server/api/settings.py", line 143, in getSettings
    return jsonify(data)
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 238, in jsonify
    indent=indent),
  File "/home/pi/octovenv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/json.py", line 126, in dumps
    rv = _json.dumps(obj, **kwargs)
  File "/usr/lib/python2.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/lib/python2.7/json/encoder.py", line 202, in encode
    chunks = list(chunks)
  File "/usr/lib/python2.7/json/encoder.py", line 427, in _iterencode
    for chunk in _iterencode_dict(o, _current_indent_level):
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 401, in _iterencode_dict
    for chunk in chunks:
  File "/usr/lib/python2.7/json/encoder.py", line 341, in _iterencode_dict
    raise ValueError("Circular reference detected")
ValueError: Circular reference detected
```

foosel, 2015-06-20:
Can't reproduce this on a vanilla RC3, need more data. Please never truncate the log file, you are depriving me of valuable context by just posting the stack trace. Especially such information like which plugins are installed and what happened before that problematic request. Also - as stated in the contribution guidelines which were linked both on the issue submission form and by GitIssueBot above - does this issue also occur without non-bundled plugins?

mattsch, 2015-06-20:
I did manage to "fix" by removing the software update bundled plugin. I'd also removed all the previously installed plugin as well as octoprint from my virtualenv.  I was getting this any time I hit the web ui.

markwal, 2015-06-20:
It probably was one of the plugins introducing a circle in the settings object via on_settings_load, so to get to the bottom of it, we'd need to know what plugins and what versions (the rest of octoprint.log that foosel mentioned) and perhaps the plugins section of config.yaml (watch for secrets you don't want on pastebin).

foosel, 2015-06-20:
Like @markwal said, it must be something specific to your runtime configuration. I run it here just fine, with the bundled softwareupdate plugin, I can't reproduce that error at all, and I know of no one else who's run into that (and this is something people would definitely notice immediately). 

We need more information, both a non-truncated log and a `config.yaml` (remove accessControl.salt, api.key and any additional secrets you might have entered, e.g. the pushbullet access token if you had the Pushbullet plugin installed).

We want to help, but we can't without that information.

foosel, 2015-06-20:
Just to make sure that plugins that do odd things with the settings that are about to be jsonified don't bring everything down with them I just added some tiny validation check, see the commit linked above.

mattsch, 2015-06-21:
Crash log:
https://gist.github.com/mattsch/eb377f1ea4ebd46de9cf
Config.yml:
https://gist.github.com/mattsch/0efce9bd23d6ee05403a

markwal, 2015-06-21:
Well that's it all right, it's those two lines in the config cause the problem and also cause foosel's new sanity check to go off:
    checks: &id001
      octoprint: *id001

If you delete them from config.yaml and restart the server, then open up settings and hit save, not only does it not hit the issue any more, but it will also output nice clean softwareupdate settings.

Now the question is, how did it get this way?  I wonder if you ran an earlier RC that wrote it this way and a later RC fixed it?  Or was RC3 your first version of 1.2?  Perhaps foosel will recognize it.

foosel, 2015-06-21:
I've never seen that, but my guess is that it was or is caused by the new config migration I had to add to fix a bug in an earlier version which caused too much data to be dumped to the config. From what I gather, pyyaml will do that stuff automatically when it detects references in the data - those shouldn't be there however.

 I'm currently not at the PC, so I can't test this myself, but it would be interesting to know if the reappears if one removed the whole softwareupdate part from the config.

markwal, 2015-06-21:
OK. I gave it a go, removing the whole software update and starting the server.  Seemed fine and it wrote a normal pretty version when I saved settings.  So it doesn't repro that easy.  I also tried going back to RC2 to repro this way and was unsuccessful.  Weird.

foosel, 2015-06-21:
It might be that it is not reproducible on RC3 anymore since it was caused by an intermediate state. 

@mattsch, did you go from RC2 directly to RC3 or did you check out some master version in between too?

foosel, 2015-06-21:
Got it. Stupid bug. Fixed and added auto-migration that should fix it where it exists.

closed 2015-06-21 by foosel

mattsch, 2015-06-21:
I'm pretty sure I pulled in an interim version from master between RC2 and RC3 to pull in the dependency link fixes (cbc6f20fa6e8e102fd6bd268cbd20b9d8c9427b7 I *think*).

foosel, 2015-06-21:
Should be fixed now in any case, thanks for reporting this

mattsch, 2015-06-21:
Interestingly, I decided to refresh my pi image with the OctoPi nightly from yesterday (2015-06-20_2015-05-05-octopi-wheezy-0.12.0.zip) and ran into the same problem.  That included the two problematic entries in the softwareupdate yaml and removing them fixed things right up.

foosel, 2015-06-21:
The nightly can't possibly have contained the fix I pushed 5 hours ago since it was built earlier than that and hence still had the bug active. Just updating the master version should have fixed it, since that would have detected the buggy configuration version and migrated it before starting up, no need to manually edit the config file.

I reproduced the problem here before applying both the fix and the migration.

mattsch, 2015-06-21:
Sure, just wanted to let you know that was there in case you got other reports from that build.

commit 33ea9c34 (2015-06-21): Fix: Software Update config could contain circular reference

Caused by a wrong variable usage. Fixed the usage and added a migration
step that fixes the circular reference if it exists. Bumping configuration
version to 2.

Fixes #946