OctoPrint/OctoPrint #947: Marlin reaches temperature. Halts print with "No Line Number with checksum" + "No Checksum with line number"

MiddleMan5, 2015-06-20:
1. What were you doing?
Printing to connected Printrbot Simple Metal
2. What did you expect to happen?
The print to start.
3. What happened instead?
Printer reaches temperature then pauses stops. Print won't continue
4. Branch & Commit or Version of OctoPrint:

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
Printrbot Simple Metal. Printrboard Rev F. Marlin
6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Google Chrome
7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

http://pastebin.com/RkBS9nM8

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

http://pastebin.com/CCzAtUGd

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

MiddleMan5, 2015-06-20:
My Serial.log file was blank

markwal, 2015-06-20:
The serial.log gets big fast so it is normally off. You need to turn it on via the serial connection settings.  But it looks like you got the relevant info from the terminal tab anyway.

From your octoprint.log, it shows you need to restart your Pi after the last plugin was installed. The new octoprint rc will not have that failure and should tell you in the UI that you need a restart (notification that pops up and disappears).  This isn't probably your problem, but thought I'd mention it.

Also, the line it is choking on looks like it is coming from your start gcode either in OctoPrint or in your slicer.  Could you paste in your whole start gcode sequence?  I think you may have a character in there that Marlin is seeing as a separator.

MiddleMan5, 2015-06-20:
I've been having this problem throughout multiple reboots, pulls, syncs, etc. This time I had not rebooted the pi in case something in the restart was causing a different (javascript error).

Heres (well part of) the "offending" Gcode files. Rendered in Simplify3D, a full and legitimate version in which I have never experienced this error before yesterday.
 
http://pastebin.com/WWDsstXs

markwal, 2015-06-20:
Well, my guess (and it is just a guess) is that if you remove those trailing colons from the G28 and G29 lines it will resolve your issue.  And if it does then I suspect that before you updated something (like OctoPrint) was stripping them out on your behalf.

foosel, 2015-06-20:
`:` is in fact a command separator. There's no valid reason why your slicer is putting it there I can think of, given that no command follows. You are basically telling it "Home and then" without specifying what else to do. Your firmware is hiccuping on that line because OctoPrint sees it as one line and prefixes it with a line number, post fixes it with a checksum and then sends it. Your firmware then splits it, and complains about the lack of a checksum at the first part and the lack of a line number at the second.

I'm not entirely sure tbh if that firmware behaviour is valid and it is something that would need to be discussed first. IF command separators are to be useful, it wouldn't really make sense to duplicate the overhead of line numbers and checksums for them IMHO. If I need to send `Nx <command1>*<checksum1>:Nx+1 <command2>*<checksum2>` instead of `Nx <command1>;<command2>*<full checksum>` I really don't gain anything from it (a line feed on the serial line and a `\n` on a serial line take both one character, so there's no advantage of using command separators if that's the way it has to be done, and only the disadvantage of having to code the firmware so that it can split on both `\n` and `:`). So that smells like a misinterpretation on the firmware side (or a very odd "works as it should").

So my personal point of view would actually be that your firmware should rather do the line number and checksum check *before* splitting. That's something that would need to be discussed with the developers of the 3d printers firmwares out there though.

I suggest you remove the `:` from your slicer's gcode (they don't make any sense anyways as no command follows them) for now. I feel reluctant to "fix" this on the host side, only to cause issues with firmware which interprets it differently. Maybe someone will write a GCODE postprocessor plugin that splits lines on ":" (which might cause other issues with `M117` commands though... that already was buggy with `;` until recently).

MiddleMan5, 2015-06-20:
Ahah! Now, I have no way of testing it, but it seems my slicer's startup code had been corrupted.
All throughout the startup, the semicolons were replaced with colons. 
(I haven't touched them, maybe someone gave a demo on my slicer?)
Anyway, that's Simplify3D's deal, apologies for the mixup.  

I don't know why I didn't catch that.

Again, this may be the fix, but I don't have a way to confirm as of right now.

foosel, 2015-06-21:
Ok. In the meantime I quickly whipped up a gcode preprocessor plugin that should take care of splitting multiple commands on a line into multiple lines with one command each upon file upload: http://plugins.octoprint.org/plugins/commandsplitter/. That should be an easy fix for anyone else stumbling over this for now.

I'll close this now. If removing the colons (or installing the plugin and uploading the files again) doesn't fix the behavior (which would surprise me a lot), feel free to comment to reopen.

closed 2015-06-21 by foosel