OctoPrint/OctoPrint #948: Using UART or GPIO to send output after printing

cegrif01, 2015-06-21:
First off, thank you for writing OctoPrint and releasing it open source, it is magnificent software!  Using the web interface, I'm trying to send a GPIO (or UART if necessary) signal to an led when the printer is finished running gcode.  Rather than going to the next set of gcode instructions automatically, I want it to wait for an input signal (button press) from the gpio to move on to the next set of gcode instructions.

I'm wondering if there's a conventional way to do this?

closed 2015-06-21 by GitIssueBot

markwal, 2015-06-21:
This seems like a good question for the Google group.

cegrif01, 2015-06-23:
@markwal I just posted the question in the Google Group: https://groups.google.com/forum/#!topic/octoprint/EvjVDymwZj4