OctoPrint/OctoPrint #950: [REQUEST] Option to disable (or pemission based) DOWNLOAD files for non logged in users?

bolsoncerrado, 2015-06-22:
I dont find quite "secure" to allow non logged in users to be able to download the files on the Octoprint environment.

Thanks!

foosel, 2015-06-22:
Just for the record, disallowing downloading of the uploaded files for anonymous users will also disable the gcode viewer. Can't have one without the other since the gcode viewer runs client side and has to download the files.

Let me also say that I'd find it a bit odd to give strangers I wouldn't trust with my files access to my OctoPrint instance. There exist solutions for these kind of problems... Making your OctoPrint instance accessible for you/your family/your hackerspace via the internet doesn't automatically mean it needs to or even should be accessible by anyone on the internet - always use VPNs for stuff like that, that's what these tools are made for!

bolsoncerrado, 2015-06-22:
Hmmmm.... I can send "friends and relatives" to my printer URL just for them to check out how their prints are coming up but they dont need to have access to my files?

Or we could set up a layer of "guest" user where they can log in and check basic features (what currently UNLOGGED features are) and for unlogged users, just keep webcam and may be "status" screens avaialble.

imrahil, 2015-06-23:
I'm pretty sure that disabling download buttons in UI is possible via simple plugin :)

foosel, 2015-06-23:
That would be security by obscurity. The download would still be possible.

imrahil, 2015-06-23:
I know. But for "friends and relatives" should be enough

MolarAmbiguity, 2015-06-23:
@bolsoncerrado which files do you want to deny access to? Timelapses? Gcode files?

bolsoncerrado, 2015-06-23:
G-code mainly

bolsoncerrado, 2015-10-05:
Can this be upgraded to "plugin":up for grabs too? :dancer:

foosel, 2015-10-05:
No, because that is a core modification with heavy implications for the general architecture, nothing you can quickly "add on".

bolsoncerrado, 2015-10-05:
:(  I was wondering then why there's a layer of "user levels" if we can't grant what can each other do or not...?

foosel, 2015-10-05:
Because I'm an evil person who does want to sleep from time to time ;)

bolsoncerrado, 2015-10-05:
BUahahahaah! :E ;) THanks! ;)

bolsoncerrado, 2016-01-22:
In the meantime, can you let me know which file and line to edit to REMOVE the DOWNLOAD FILE button from the File Manager?

Many thanks!

foosel, 2016-01-22:
That is supposed to accomplish what exactly? It will still be downloadable
by anyone. That's like leaving your car parked unlocked and hanging a sign
into the window: do not open.

But if you like to have the illusion of security:
src/octoprint/templates/sidebar/files.jinja2

bolsoncerrado, 2016-01-22:
FOr those that are totally NEW to octoprint or not "tech" savy, they should not find it "easy" to download any .gcode on my SDcard, right?

bolsoncerrado, 2016-01-22:
OR if instead of removal, why not just "DISABLE" the icon like when someone is not logged in as admin cant just delete a file?

bolsoncerrado, 2016-01-25:
No?

foosel, 2016-01-25:
You keep on suggesting snake-oil "solutions". 

I'm not going to add simulated security into OctoPrint and then have people open bugs (rightly so) that their stuff is still accessible even though a button is disabled or a icon is not included somewhere. With #1110 there is a ticket of which the solution will also *properly* solve this here (and a couple other tickets in the system). Doing it properly however is a lot of work and I currently do not have the time to get into that.

If you want to simulate a proper permission based approach by removing buttons, I told you where to find the file above. It is open source after all, you can adjust it yourself if you are not happy with what you find.

foosel, 2016-01-25:
Speaking of which, closing this since it is now tracked as part of #1110.

closed 2016-01-25 by foosel

bolsoncerrado, 2016-01-25:
Since im not a programmer im definetly missing something here...

You keep saying I want you to add "open bugs" to your code. Just please reply this question: when someone types an octoprint URL and is NOT logged in as admin, some icons on the file manager, like the DELETE button, are grayed out. The DOWNLOAD button is not. I'm asking you if you can GRAY it out when not logged in as admin and you claim anyone can still download the files. Anyone? Or anyone "tech savy"/with enough computer/programming knowleadge?

I'm NOT here to bug you, I LOVE your tool! I'm just trying to figure it out! Nothing else! And OF COURSE! Knowing your system and the "open doors" to anyone I share the URL with! So I appreciate A LOT the time you put replying me to these questions! THank you!

foosel, 2016-01-25:
Graying out a button doesn't disable the endpoint on the server. So anyone who is able to read the [docs](http://docs.octoprint.org/en/master/api/fileops.html#retrieve-a-specific-file-s-information) or who just happens to hit F12 and takes a look at e.g. how the GCODE viewer is fetching its data will get the URL to download the file (btw - if done properly, no file download also means no GCODE viewer).

The problem with doing that security-by-obscurity approach you are suggesting basically by "just disabling the button" is - yes, it will prevent people who don't know much. But your problem aren't those people. You don't want security that only works if the user is - sorry - stupid enough for it. That is not the way to do software development. Something you develop has to work reliably regardless who's on the other side of the keyboard. And that is especially important when it comes to security. You never design for the weakest link in the chain, but for the script kiddie which desperately wants nothing more than to break and steal your stuff.

Now, consider that I did what you asked here and added an option "disable download button for anonymous users". That would be a feature that not only you - who would hopefully be aware of the problems with that now - would use, but that would probably also be used by others, who wouldn't be aware of the issues with that and potentially rely on it securing things that wouldn't really be secured. And that would lead to bugs being reported - rightly so, as I said.

It just doesn't make sense implemented half-hearted "look the other way please" measures. The proper way to do it is to introduce a more granular permission system that controls access to each and every aspect of OctoPrint, as outlined in #1110. But that's a huge change that I will probably not be able to do for 1.3.0, since it is very disruptive.

nophead, 2016-01-25:
All anybody has to do to download a file is type this into the browser nav
bar: http://octopi.local/downloads/files/local/yourfile.cgo. Anybody can
see this by hovering over a button on a non-nobbled version of OctoPrint.
Removing the button doesn't stop that.

foosel, 2016-01-25:
I actually thought too complicated and @nophead 's point is even better, hovering over a button in some other OctoPrint instance or whatever would already be enough to find the URL.

bolsoncerrado, 2016-01-25:
OK i think i follow you all now... so is there any "drastic" solution (even if it's just "for me" right now) to cut the download feature somehow until you develop the user levels API? or it comes "bundled" with the Gcode viewer "yes or yes"?

foosel, 2016-01-26:
No download, no GCODE viewer. The browser has to get the file in order to be able to display it and if the browser can get the file, so can the user.

The "drastic" solution would be to remove the download endpoint declaration from your sources. That would be [this line](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/__init__.py#L359). That will make the download button stop working. To remove the button, delete [this line](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/templates/sidebar/files.jinja2#L14).

Keep in mind that this will probably cause problems for you when updating. 

I'm not sure that keeping people from downloading GCODE files, which are useless for any but my own printer usually anyways, would be worth that for me. But YMMV.

bolsoncerrado, 2016-01-26:
Thank you Sir!
Well since OP allows for STL upload too, sometimes I just dont wanna share stuff I may work with NDAs. Even if its GCODE that could render something pretty much usable... even if it's not 100% perfect.

foosel, 2016-01-26:
> Sir

err.. no ;)