OctoPrint/OctoPrint #952: [request] save print data to log

luotinen, 2015-06-22:
After a print finishes, could we save the following print information to a log file:
- printed file name
- printed file location (to link directly to the files from the log)
- print succesful or failed
- estimated print time
- measured print time
- (new text variable) filament type if set by user, otherwise empty
- filament length meters
- filament volume cubic centimeters
- (new text variable) custom comment max. 255 characters (for versioning)

This would enable viewing log files and creating statistics of:
- print history with links to files (for easy reprinting)
- actual print times
- filament consumption

markwal, 2015-06-22:
There's a plugin for this.  If you update to version 1.2.0-RC3 and look at the Plugin Manager in Settings, you can install "Print History" that does a lot of that.  You could post this request to that plugin manager, enhance it and give a PR to the author or write a different plugin.  :-)

foosel, 2015-06-30:
Didn't see that post by @markwal before, he's of course completely right.

closed 2015-06-30 by foosel