OctoPrint/OctoPrint #954: [Request] In gcode viewer center and zoom default ?

manu69x, 2015-06-23:
I have a delta printer
in gcodeviewer folder there is a way to preset
center and zoom default ?

foosel, 2015-06-23:
Yes, look in the settings, printer parameters (or printer profiles on 1.2.0), bed center. Has been there for a long time. The Settings dialog is your friend.

closed 2015-06-23 by foosel

manu69x, 2015-06-23:
I have Octopi with Octoprint Version: 1.1.1-30-g4fede5a (master branch)
Can bee that I have old version ?

foosel, 2015-06-23:
Indeed looks like it, although I could have sworn that was already available on 1.1.1. But 1.2.0 is about to be released. Use that then.

imrahil, 2015-06-23:
+1 for that - there is no option to set this values to true by default...
@foosel "bed center" is for machine, not for gcodeviewer

foosel, 2015-06-23:
@imrahil guess what the gcode viewer is evaluating to decide on where to render the origin, which the OP was also asking about...

And for allowing to store the gcode viewer settings, there already is a ticket for that, use the search, people, that's what it's there for ;)