OctoPrint/OctoPrint #956: OctoPrint crashing after adding any slicing profile

mheilimo, 2015-06-27:
Hi!
I used latest version of OctoPrint and when I try to add default Slic3r slicer profile my Octoprint crashing and give this error:

2015-06-28 00:14:10,870 - octoprint.server - CRITICAL - Now that is embarrassing... Something really really went wrong here. Please report this including the stacktrace below in OctoPrint's bugtracker. Thanks!
2015-06-28 00:14:10,872 - octoprint.server - ERROR - Stacktrace follows:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/__init__.py", line 451, in run
    ioloop.start()
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/ioloop.py", line 780, in start
    self._run_callback(callback)
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/ioloop.py", line 565, in _run_callback
    ret = callback()
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/stack_context.py", line 275, in null_wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/gen.py", line 697, in <lambda>
    self.future, lambda f: self.run())
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/gen.py", line 622, in run
    yielded = self.gen.send(value)
  File "/usr/local/lib/python2.7/dist-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/web.py", line 1332, in _execute
    result = method(*self.path_args, **self.path_kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/util/tornado.py", line 405, in _handle_method
    self._fallback(self.request, body)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/util/tornado.py", line 461, in __call__
    WsgiInputContainer.environ(request, body), start_response)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/util/__init__.py", line 219, in __call__
    return self.app(environ, start_response)
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/util/flask.py", line 363, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/server/api/slicing.py", line 109, in slicingPatchSlicerProfile
    profile = slicingManager.load_profile(slicer, name)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/slicing/__init__.py", line 349, in load_profile
    return self._load_profile_from_path(slicer, path, require_configured=require_configured)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/slicing/__init__.py", line 554, in _load_profile_from_path
    return self.get_slicer(slicer, require_configured=require_configured).get_slicer_profile(path)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/slicing/__init__.py", line 187, in get_slicer
    raise SlicerNotConfigured(slicer)
SlicerNotConfigured

So this happen when I was clicking star that define 'make default' - what went wrong? Nicer error at least would be nice.

mheilimo, 2015-06-27:
My slic4r is also latest 1.2.9

mheilimo, 2015-06-27:
well now it follow guidance....

foosel, 2015-06-27:
Wait... you've been trying to add a slic3r profile to the "CuraEngine" settings dialog?

foosel, 2015-06-27:
Also...

> Link to octoprint.log on gist.github.com or pastebin.com (ALWAYS INCLUDE AND **DO NOT TRUNCATE**):

The *full* log please

mheilimo, 2015-06-27:
No this is Slic3r - I disabled my Cura engine because it crashing same way. That reason I try now to add slic3r - there is my profile: http://1drv.ms/1LO2aJx

mheilimo, 2015-06-27:
There is my full dialog that you can see that it is right place: 
http://1drv.ms/1LO2mbF

foosel, 2015-06-27:
That plugin is not published yet for a reason. It is not ready for general consumption or it would be listed on the plugin repository. Uninstall it please. Then reproduce with the bundled CuraEngine plugin and provide a **full** log file (the whole of `octoprint.log`).

mheilimo, 2015-06-27:
There is also full log and there you will find other problem:
http://1drv.ms/1QWa3mG
if you have to extruder it can't import them correctly:
ValueError: invalid literal for float(): 0.4,0.4
2015-06-27 23:52:31,647 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for retract_before_travel from an imported profile, using default
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_slic3r/profile.py", line 250, in convert_value
    return int(value)
ValueError: invalid literal for int() with base 10: '2,2'
2015-06-27 23:52:31,649 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for retract_length from an imported profile, using default
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_slic3r/profile.py", line 250, in convert_value
    return int(value)
ValueError: invalid literal for int() with base 10: '1,1'
2015-06-27 23:52:31,650 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for retract_length_toolchange from an imported profile, using default
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_slic3r/profile.py", line 250, in convert_value
    return int(value)
ValueError: invalid literal for int() with base 10: '10,10'
2015-06-27 23:52:31,652 - plugins.slic3r.slic3r.profile - ERROR - Got an exception while trying to convert the value for retract_lift from an imported profile, using default

foosel, 2015-06-27:
Again. Please uninstall the *not yet released Slic3r plugin that is still under development*. It is not released for a reason. It is not bundled for a reason. Uninstall it. Provide the log of the error you are getting with the *bundled CuraEngine plugin*. The Slic3r plugin is *not supported here*.

We can spent the whole night running in circles here, or you can show me the problem you are having with the CuraEngine plugin and we can try to solve that instead.

mheilimo, 2015-06-27:
Take it easy... How I can know that this don't work at all with slic3r because there is no note that it is very unstable (yes it is not released, but I was expecting that it works someway...)

there is my new log with cura: 
http://1drv.ms/1QWa3mG

and there is my cura profile:
http://1drv.ms/1ICeOrF

Some reason I can't anymore even add this profile there and crashing to error: 
2015-06-28 01:03:03,029 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.54
2015-06-28 01:03:05,040 - octoprint.plugins.cura - INFO - Path to CuraEngine has not been configured yet or does not exist (currently set to '~/cura_engine'), Cura will not be selectable for slicing
2015-06-28 01:03:05,781 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.54
2015-06-28 01:03:43,011 - octoprint.plugin.cura.profile - ERROR - Error while reading profile INI file from /tmp/octoprint-file-upload-0wL_Qf.tmp
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.2.0-py2.7.egg/octoprint/plugins/cura/profile.py", line 360, in from_cura_ini
    config.read(path)
  File "/usr/lib/python2.7/ConfigParser.py", line 305, in read
    self._read(fp, filename)
  File "/usr/lib/python2.7/ConfigParser.py", line 546, in _read
    raise e
ParsingError: File contains parsing errors: /tmp/octoprint-file-upload-0wL_Qf.tmp
        [line 36]: 'G28 ; Home extruder\r\n'
        [line 37]: 'G1 Z15 F{Z_TRAVEL_SPEED}\r\n'
        [line 38]: 'M107 ; Turn off fan\r\n'
        [line 39]: 'G90 ; Absolute positioning\r\n'
        [line 40]: 'M82 ; Extruder in absolute mode\r\n'
        [line 41]: '{IF_BED}M190 S{BED}\r\n'
        [line 43]: '{IF_EXT0}M104 T0 S{TEMP0}\r\n'
        [line 44]: '{IF_EXT1}M104 T1 S{TEMP1}\r\n'
        [line 45]: 'G92 E0 ; Reset extruder position\r\n'
        [line 47]: '{IF_EXT0}M109 T0 S{TEMP0}\r\n'
        [line 48]: '{IF_EXT1}M109 T1 S{TEMP1}\r\n'
        [line 49]: '"""\r\n'
        [line 53]: 'M107 ; Turn off fan\r\n'
        [line 55]: 'G91 ; Relative positioning\r\n'
        [line 56]: '{IF_EXT0}T0\r\n'
        [line 57]: '{IF_EXT0}G1 E-1 ; Reduce filament pressure\r\n'
        [line 58]: 'M104 T0 S0\r\n'
        [line 59]: '{IF_EXT1}T1\r\n'
        [line 60]: '{IF_EXT1}G1 E-1 ; Reduce filament pressure\r\n'
        [line 61]: 'M104 T1 S0\r\n'
        [line 62]: 'G90 ; Absolute positioning\r\n'
        [line 63]: 'G92 E0 ; Reset extruder position\r\n'
        [line 64]: 'M140 S0 ; Disable heated bed\r\n'
        [line 65]: 'M84 ; Turn steppers off\r\n'
        [line 66]: '"""\r\n'
2015-06-28 01:03:43,018 - octoprint.plugins.cura - WARNING - Could not convert profile, aborting

mheilimo, 2015-06-27:
And cura says that if I run it manually:
Cura_SteamEngine version 14.12.1
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Default config 'default.cfg' not used

mheilimo, 2015-06-27:
and when you get angry about my comments please understand that I didn't see you comment that it don't work with slic3r before I posted these other informations because I didn't click re-refesh when I writing to you. Thanks!

foosel, 2015-06-27:
Ok, I understand. Please try to understand though that I get a lot of tickets that start out with people ignoring the big box that says "Please review the guidelines for contributing to this repository", which also includes the bug reporting guide, which at the very top also includes things like "This is the bug tracker for OctoPrint, not OctoPi or any [non bundled] OctoPrint plugins" and "Please disable all third party plugins and try to reproduce". Can get a bit tiring ;)

----

That `curaSettings.rcp` file you linked does not look like a valid Cura profile. Cura profiles should be valid ini files, and I've also never seen that extension associated with them before. That `startGCcode` in your profile there for example is not valid for ini files, hence the error when trying to import it. See an example for something that OctoPrint can import [here](https://gist.github.com/foosel/3527ebbe21928eb65f43). 

Can you please tell me where/how you got this profile? What was the exact Cura version from which that was exported?

mheilimo, 2015-06-27:
In understand your problems well - I working myself in software business :)

I am using repetier 1.0.6 and cura engine is version Cura_SteamEngine version 14.12. I installed new cura engine that I can get profile in ini format and my log after clicking same 'make default' star (and after crash):
http://1drv.ms/1QWa3mG

and my ini based profile looks like that:
http://1drv.ms/1Hnau34

for me it looks that there is no error in import anymore - only after trying to get that to 'make default'.

foosel, 2015-06-27:
Ok, now we are getting something :) I just could reproduce your problem, happens when you try to make some profile a default profile but have not yet configured the path to CuraEngine. That's a bug, will look into it.

mheilimo, 2015-06-27:
Thanks!
My cura engine is configured if I run that call manually it says: 
Cura_SteamEngine version 14.12.1
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program. If not, see http://www.gnu.org/licenses/.
Default config 'default.cfg' not used

And my cura path is: ~/cura_engine - there is just one file what is that binary engine file. Is that right?

foosel, 2015-06-27:
2015-06-28 01:41:24,766 - octoprint.plugins.cura - INFO - Path to CuraEngine has not been configured yet or does not exist (currently set to '~/cura_engine'), Cura will not be selectable for slicing

You need to specify the full path in OctoPrint, not just `~/cura_engine`. `~` is something that the command line will extend to `/home/<youruser>`, but OctoPrint expects an absolute path. Try `/home/pi/cura_engine` instead.

To get your setup running again you can edit your `/home/pi/.octoprint/config.yaml` file and set the occurence of "~/cura_engine" to `/home/pi/cura_engine`, then restart.

closed 2015-06-30 by foosel

mheilimo, 2015-07-08:
Thanks this works - I propose to fix also error message that says that ~/... path don't work . This was in many instructions. It should say something that 'can't use relative paths or only absolute path allowed.

foosel, 2015-07-08:
> This was in many instructions

Anything about that in the wiki/the official docs? If so that should be fixed...

foosel, 2015-07-08:
Did something better... First of all, ~ paths work now, and I added a "Test" button:

[image]

[image]

`devel` branch, will be in 1.3.0

mheilimo, 2015-07-08:
Thank you - great solution! I love that product :)

mheilimo, 2015-07-08:
Sorry about reply delay originally for that path issue - I spend time to try to find guidance - I am quite sure that it was official example, but I have to double check it before I can guarantee that (I could find link)- I will try to do that in this week.

commit 13ac06c6 (2015-06-28): Fix: Slicing profiles now can always be set as default

Previously depended on the slicer being already configured, which is not
really necessary.

Closes #956
commit 4b110325 (2015-06-28): Fix: Slicing profiles now can always be set as default

Previously depended on the slicer being already configured, which is not
really necessary.

Closes #956
(cherry picked from commit 13ac06c)